ros-perception/perception_pcl #71: pointcloud_to_laserscan 1.2.4 broken

cottsay, 2015-01-17:
`pointcloud_to_laserscan` seems to have been broken by https://github.com/ros-perception/perception_pcl/commit/331cd9c3596ce241865b09cc7c20b9cbb5bb9318 [1] on Fedora.

I can't really figure out how it is working on Ubuntu but not Fedora. I'd be more than willing to test fixes, or even supply a Fedora machine for you to test on, but I'm at my wit's end trying to find a solution...

This is blocking a pretty fair amount of packages from building on Fedora...

Thanks,

--scott

[1] http://csc.mcs.sdsmt.edu/jenkins/job/ros-indigo-pointcloud-to-laserscan_binaryrpm_heisenbug_x86_64/21/consoleFull

wjwwood, 2015-01-17:
@cottsay if you can give me ssh to a build slave-like machine, I'll give it a go.

wjwwood, 2015-01-17:
I can reproduce it on my mac laptop:

```
[pointcloud_to_laserscan] ==> '/private/tmp/p2l_ws/build/pointcloud_to_laserscan/build_env.sh /usr/bin/make -j4 -l4' in '/private/tmp/p2l_ws/build/pointcloud_to_laserscan'
Scanning dependencies of target pointcloud_to_laserscan
[ 50%] Building CXX object CMakeFiles/pointcloud_to_laserscan.dir/src/pointcloud_to_laserscan_nodelet.cpp.o
Linking CXX shared library /tmp/p2l_ws/devel/lib/libpointcloud_to_laserscan.dylib
Undefined symbols for architecture x86_64:
  "std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const& tf2::getFrameId<geometry_msgs::Point32_<std::__1::allocator<void> > >(geometry_msgs::Point32_<std::__1::allocator<void> > const&)", referenced from:
      geometry_msgs::Point32_<std::__1::allocator<void> >& tf2_ros::BufferInterface::transform<geometry_msgs::Point32_<std::__1::allocator<void> > >(geometry_msgs::Point32_<std::__1::allocator<void> > const&, geometry_msgs::Point32_<std::__1::allocator<void> >&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, ros::Time const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, ros::Duration) const in pointcloud_to_laserscan_nodelet.cpp.o
  "void tf2::doTransform<geometry_msgs::Point32_<std::__1::allocator<void> > >(geometry_msgs::Point32_<std::__1::allocator<void> > const&, geometry_msgs::Point32_<std::__1::allocator<void> >&, geometry_msgs::TransformStamped_<std::__1::allocator<void> > const&)", referenced from:
      geometry_msgs::Point32_<std::__1::allocator<void> >& tf2_ros::BufferInterface::transform<geometry_msgs::Point32_<std::__1::allocator<void> > >(geometry_msgs::Point32_<std::__1::allocator<void> > const&, geometry_msgs::Point32_<std::__1::allocator<void> >&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, ros::Time const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, ros::Duration) const in pointcloud_to_laserscan_nodelet.cpp.o
  "ros::Time const& tf2::getTimestamp<geometry_msgs::Point32_<std::__1::allocator<void> > >(geometry_msgs::Point32_<std::__1::allocator<void> > const&)", referenced from:
      geometry_msgs::Point32_<std::__1::allocator<void> >& tf2_ros::BufferInterface::transform<geometry_msgs::Point32_<std::__1::allocator<void> > >(geometry_msgs::Point32_<std::__1::allocator<void> > const&, geometry_msgs::Point32_<std::__1::allocator<void> >&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, ros::Time const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, ros::Duration) const in pointcloud_to_laserscan_nodelet.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/tmp/p2l_ws/devel/lib/libpointcloud_to_laserscan.dylib] Error 1
make[1]: *** [CMakeFiles/pointcloud_to_laserscan.dir/all] Error 2
make: *** [all] Error 2
[pointcloud_to_laserscan] <== '/private/tmp/p2l_ws/build/pointcloud_to_laserscan/build_env.sh /usr/bin/make -j4 -l4' failed with return code '2'
```

I believe this is the case because OS X and Fedora have the same default linker behavior, which is to error if symbols are undefined at linkedit time. I think this only makes sense, but Ubuntu and other Linux's tend to silently pass on missing symbols hoping they are found at runtime. I'm looking for a better solution.

paulbovbel, 2015-01-18:
Thanks for looking at this, is it something I'm linking incorrectly?

wjwwood, 2015-01-18:
I don't think so, I haven't yet figured out the issue, but I think it may be something wrong with how the tf2 headers are setup. I'm still looking into it, but no luck yet.

cottsay, 2015-01-20:
I still don't know what caused this, but (https://github.com/ros-perception/perception_pcl/commit/aef478e80911a7d540d41e171cac778d0fa28c85) removes references to `Point32` and avoids the problem altogether.

Builds look good now: http://csc.mcs.sdsmt.edu/jenkins/job/ros-indigo-pointcloud-to-laserscan_binaryrpm_21_x86_64/18/consoleFull

@wjwwood - unless you want to continue investigating why this happened, this issue may be closed.

wjwwood, 2015-01-20:
Hmm, seems to work for me on `indigo-devel`, so I guess that's fine to close this (I'll let one of you do that when you're satisfied).

I was on a similar track, but I'm still not entirely sure why it was failing. Anyways I'm always glad to drop something off my plate.

paulbovbel, 2015-01-21:
I think it was my fault. doTransform and related aren't actually implemented for geometry_msgs::Point32.

closed 2015-01-21 by paulbovbel