ros-perception/perception_pcl #73: pointcloud_to_laserscan package not working in ROS indigo

nrkumar93, 2015-02-01:
The pointcloud_to_laserscan package is not working properly in ROS Indigo. The Indigo build is done recently and seems to contain some error. As required by the node I have published the pointcloud data from Kinect in PointCloud2 format in /cloud_in topic. But the laserscan data that must be produced and published by Kinect in /scan topic seems erroneous. It returns an array full of 'Inf'. Please fix the error.

My code for publishing in /cloud_in is as follows,
```python
#!/usr/bin/env python
import roslib
import rospy
import geometry_msgs.msg
from sensor_msgs.msg import PointCloud2

def callback(data):
	pub = rospy.Publisher('/cloud_in', PointCloud2,queue_size=10)
	pub.publish(data)
	
def pcl_2D():
	rospy.init_node('pcl_2D', anonymous=True)
	global listener	
	rospy.Subscriber("/camera/depth/points", PointCloud2, callback)
	rospy.spin()

if __name__=='__main__':
	pcl_2D()
	
```

paulbovbel, 2015-02-01:
Take a look at topic remapping in https://github.com/ros-perception/perception_pcl/blob/indigo-devel/pointcloud_to_laserscan/launch/sample_node.launch

I've recently tested on my setup and everything was working fine. Try compiling from source, and looking at the DEBUG messages to see if/why points are getting filtered out.

If you still can't find the source of the problem, please get a bag of the Kinect data, and post a link here, along with a (gist) roslaunch configuration that mirrors your pointcloud_to_laserscan setup.

Thanks for bringing this up, and keep me posted!

nrkumar93, 2015-02-06:
I tried all the methods u stated and failed continuously. When installed using apt-get I found the node 'pointcloud_to_laserscan' subscribing to /tf and /tf_static topic. I tried all the possible combinations for launch files (both simple_node and simple_nodelet) that suit my system. No combination seemed to work. But I didn't have any hope to work as it subscribed to /tf and /tf_static of 'tfMessafe' type rather than a sensor_msgs/PointCloud2 type. 

So I installed from source as u suggested. When I ran the launch file (both simple_node and simple_nodelet) I got this error when i tweaked simple_node 
>[ERROR] [1423263697.747497985]: Failed to load nodelet [/pointcloud_to_laserscan] of type [pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet]: Could not find library corresponding to plugin pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

and the following when I tweaked simple_nodelet
>ERROR: cannot launch node of type [pointcloud_to_laserscan/nodelet]: can't locate node [nodelet] in package [pointcloud_to_laserscan]

But the alterations I did were correct to my knowlege. 

After this I ended up running the node alone using the package without using the launch file. Then I found from 'rosnode info' that it publishes in /scan and subscribes to /cloud_in from ros wiki. I ran my intermediatary Python node that published in /cloud_in and returned only 'inf' in ranges field of /scan topic.

The bag file for your reference is given the bottom link. It has been compressed which can be decompressed using 
>rosbag decompress *.bag

Please make the node working. I mean independent of the launch file also just like any other package in ROS.

Thanks in advance. :+1: 

Bag file: https://drive.google.com/file/d/0B7uPJVMkbpRKMEFWR3FzMDVMQ1U/view?usp=sharing

paulbovbel, 2015-02-18:
So, I checked your bag file, and the node worked as expected. A scan was output, with all components Inf. Turning on debug logging for the node showed why (lots of nan/out of range points).

You're also using a python node to subscribe and publish the cloud points, instead of <remap/> tags in the launch file.

Please go through http://wiki.ros.org/ROS/Tutorials/UsingRqtconsoleRoslaunch.

closed 2015-02-18 by paulbovbel