ros-perception/perception_pcl #84: This may or may not be an issue I am unable to find a solution

rnunziata, 2015-03-23:
I am running pointcloud_to_laserscan . When I display the pointcloud2 in rviz after rotation the camera frame is displays correctly along the rviz ground plane. However the laser from pointcloud_to_laserscan produces a scan line that is perpendicular to the ground ignoring the rotation. So I get a vertical scan line instead of a horizontal parallel to the ground. From what I can see it does not subscribe to tf or camera_info. I am using the gazebo kenitic plugin in simulation.

A extended conversation of trying to fix this issue is found here along with images and data. I hope this is ok there is a fair amount to data and back and forth dialog on trying various solutions. 
http://answers.ros.org/question/205552/pointcloud_to_laserscan-rotation-of-scan-line/

LucidOne, 2015-05-27:
Do you have a launch file? Error logs?
Have you tried running roslaunch with the --screen option?

SteveMacenski, 2019-10-16:
Inactive, closing

closed 2019-10-16 by SteveMacenski