ros-perception/perception_pcl #103: Aborted (core dumped)

superstage2015, 2015-06-30:
Hi,
I am using pointcloud_to_laserscan node, but when I try to rosrun it, i got a Aborted (core dumped) just after starting the node.
The output his never the same, there some examples :

[ INFO] [1435670694.663510226]: Initializing nodelet with 8 worker threads.
*** Error in `/opt/ros/indigo/lib/pointcloud_to_laserscan/pointcloud_to_laserscan_node': free(): invalid next size (normal): 0x0000000001bd61c0 ***
Aborted (core dumped)

[ INFO] [1435670685.238742355]: Initializing nodelet with 8 worker threads.
*** Error in `/opt/ros/indigo/lib/pointcloud_to_laserscan/pointcloud_to_laserscan_node': corrupted double-linked list: 0x00000000025d4ed0 ***
Aborted (core dumped)

[ INFO] [1435670683.980478159]: Initializing nodelet with 8 worker threads.
*** Error in `/opt/ros/indigo/lib/pointcloud_to_laserscan/pointcloud_to_laserscan_node': malloc(): memory corruption: 0x000000000068e180 ***
Aborted (core dumped)

It's very strange, because sometime the node will run without issues. Maybe there isn't enough memory left in my systems. I also sometime got Segmentation fault (core dumped).

Have you run into some similar issues ?

Context : I using a kinect v1 with openni_camera with this drivers : https://github.com/avin2/SensorKinect.

paulbovbel, 2015-08-03:
Sorry for the delay. This is very strange, could you please provide a launch configuration/bag file to reproduce this?

SteveMacenski, 2019-10-16:
Inactive, closing. Looks like user compilation issue not a package problem

closed 2019-10-16 by SteveMacenski