ros-perception/perception_pcl #104: OS X build issue

mikepurvis, 2015-07-17:
Not sure what exactly is going on here, but this worked a few months ago and is broken now:

```
[pcl_ros] ==> '/Users/test/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/local/bin/cmake /Users/test/indigo_desktop_full_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/test/indigo_desktop_full_ws/devel -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/include/python2.7' in '/Users/test/indigo_desktop_full_ws/build/pcl_ros'
-- The C compiler identification is AppleClang 6.1.0.6020053
-- The CXX compiler identification is AppleClang 6.1.0.6020053
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.28") 
-- checking for module 'eigen3'
--   found eigen3, version 3.2.4
-- Found Eigen: /usr/local/Cellar/eigen/3.2.4/include/eigen3  
-- Eigen found (include: /usr/local/Cellar/eigen/3.2.4/include/eigen3)
-- Found eigen: /usr/local/Cellar/eigen/3.2.4/include/eigen3  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'libopenni'
--   package 'libopenni' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** io features related to openni will be disabled
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** io features related to openni2 will be disabled
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- Found libusb-1.0: /usr/local/include  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/local/Cellar/flann/1.8.4/lib/libflann_cpp_s.a  
-- checking for module 'libopenni'
--   package 'libopenni' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** visualization features related to openni will be disabled
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** visualization features related to openni2 will be disabled
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- Found qhull: /usr/local/lib/libqhull_p.dylib  
-- checking for module 'libopenni'
--   package 'libopenni' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** apps features related to openni will be disabled
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework  
CMake Error at /usr/local/share/pcl-1.8/PCLConfig.cmake:49 (message):
  simulation is required but glew was not found
Call Stack (most recent call first):
  /usr/local/share/pcl-1.8/PCLConfig.cmake:566 (pcl_report_not_found)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:729 (find_external_library)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/test/indigo_desktop_full_ws/build/pcl_ros/CMakeFiles/CMakeOutput.log".
[pcl_ros] <== '/Users/test/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/local/bin/cmake /Users/test/indigo_desktop_full_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/test/indigo_desktop_full_ws/devel -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/include/python2.7' failed with return code '1'
                                                                                                                                                                      
Failed   <== pcl_ros                          [ 5.5 seconds ]  
```

This is against PCL 1.7.1, from the deps tap: https://github.com/ros/homebrew-deps/blob/master/pcl.rb

@wjwwood, @nikolausdemmel, any idea what happened here?

wjwwood, 2015-07-17:
Based on your output that looks like PCL 1.8.

mikepurvis, 2015-07-17:
Hmm. Looks like ros/deps/pcl is 1.7.1, and homebrew/science/pcl is 1.7.2, but the version that's working on my main computer is HEAD from around March 2015.

Currently experimenting; may submit a PR to update homebrew-deps.

mikepurvis, 2015-07-17:
This fixes it: https://github.com/ros/homebrew-deps/pull/9

wjwwood, 2015-07-21:
Since you closed ros/homebrew-deps#9, I'm guessing this still isn't resolved?

mikepurvis, 2015-07-21:
Sorry, never updated here.

The issue resolves by using the PCL 1.7.2 from homebrew/science/pcl.

closed 2015-07-21 by mikepurvis