ros-perception/perception_pcl #106: Eigen3 cmake module not found.

rcodddow, 2015-07-30:
I am try to install Jade on OS X 10.10 Yosemite. 

I am stuck trying to compile plc_ros because Eigen3 can be found

==> Processing catkin package: 'pcl_ros'
==> Building with env: '/Users/robert/ros_catkin_ws/install_isolated/env.sh'
==> cmake /Users/robert/ros_catkin_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/robert/ros_catkin_ws/devel_isolated/pcl_ros -DCMAKE_INSTALL_PREFIX=/Users/robert/ros_catkin_ws/install_isolated -DCMAKE_BUILD_TYPE=Release -G Unix Makefiles in '/Users/robert/ros_catkin_ws/build_isolated/pcl_ros'
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/Users/robert/ros_catkin_ws/build_isolated/pcl_ros/CMakeFiles/CMakeOutput.log".
<== Failed to process package 'pcl_ros': 
  Command '['/Users/robert/ros_catkin_ws/install_isolated/env.sh', 'cmake', '/Users/robert/ros_catkin_ws/src/perception_pcl/pcl_ros', '-DCATKIN_DEVEL_PREFIX=/Users/robert/ros_catkin_ws/devel_isolated/pcl_ros', '-DCMAKE_INSTALL_PREFIX=/Users/robert/ros_catkin_ws/install_isolated', '-DCMAKE_BUILD_TYPE=Release', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /Users/robert/ros_catkin_ws/build_isolated/pcl_ros && /Users/robert/ros_catkin_ws/install_isolated/env.sh cmake /Users/robert/ros_catkin_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/robert/ros_catkin_ws/devel_isolated/pcl_ros -DCMAKE_INSTALL_PREFIX=/Users/robert/ros_catkin_ws/install_isolated -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'

Command failed, exiting.

--------------------------------------

If I edit the CMakeList.txt  to require the Eigen package instead of Eigen3 this is what i get. I have tried the migration guide spec aswell http://wiki.ros.org/jade/Migration#Eigen_CMake_Module_in_cmake_modules. No change.

==> Processing catkin package: 'pcl_ros'
==> Building with env: '/Users/robert/ros_catkin_ws/install_isolated/env.sh'
==> cmake /Users/robert/ros_catkin_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/robert/ros_catkin_ws/devel_isolated/pcl_ros -DCMAKE_INSTALL_PREFIX=/Users/robert/ros_catkin_ws/install_isolated -DCMAKE_BUILD_TYPE=Release -G Unix Makefiles in '/Users/robert/ros_catkin_ws/build_isolated/pcl_ros'
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
CMake Warning at /Users/robert/ros_catkin_ws/install_isolated/share/cmake_modules/cmake/Modules/FindEigen.cmake:62 (message):
  The FindEigen.cmake Module in the cmake_modules package is deprecated.

  Please use the FindEigen3.cmake Module provided with Eigen.  Change
  instances of find_package(Eigen) to find_package(Eigen3).  Check the
  FindEigen3.cmake Module for the resulting CMake variable names.

Call Stack (most recent call first):
  CMakeLists.txt:7 (find_package)

-- Eigen found (include: /usr/local/Cellar/eigen/3.2.4/include/eigen3)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'libopenni'
--   package 'libopenni' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** io features related to openni will be disabled
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** io features related to openni2 will be disabled
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- checking for module 'libopenni'
--   package 'libopenni' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** visualization features related to openni will be disabled
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** visualization features related to openni2 will be disabled
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- checking for module 'libopenni'
--   package 'libopenni' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** apps features related to openni will be disabled
CMake Error at /usr/local/share/pcl-1.8/PCLConfig.cmake:49 (message):
  simulation is required but glew was not found
Call Stack (most recent call first):
  /usr/local/share/pcl-1.8/PCLConfig.cmake:566 (pcl_report_not_found)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:729 (find_external_library)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/robert/ros_catkin_ws/build_isolated/pcl_ros/CMakeFiles/CMakeOutput.log".
<== Failed to process package 'pcl_ros': 
  Command '['/Users/robert/ros_catkin_ws/install_isolated/env.sh', 'cmake', '/Users/robert/ros_catkin_ws/src/perception_pcl/pcl_ros', '-DCATKIN_DEVEL_PREFIX=/Users/robert/ros_catkin_ws/devel_isolated/pcl_ros', '-DCMAKE_INSTALL_PREFIX=/Users/robert/ros_catkin_ws/install_isolated', '-DCMAKE_BUILD_TYPE=Release', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /Users/robert/ros_catkin_ws/build_isolated/pcl_ros && /Users/robert/ros_catkin_ws/install_isolated/env.sh cmake /Users/robert/ros_catkin_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/robert/ros_catkin_ws/devel_isolated/pcl_ros -DCMAKE_INSTALL_PREFIX=/Users/robert/ros_catkin_ws/install_isolated -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'

Command failed, exiting.

paulbovbel, 2015-08-03:
So this seems like two separate issues. I'm not too familiar with OSX so bear with me. For your openni issue, take a look at https://github.com/ros-perception/perception_pcl/issues/104.

For the Eigen cmake issue, the FindEigen3.cmake module comes with cmake version 2.8 (at least). What version of cmake are you using?

wjwwood, 2015-08-03:
My CMake doesn't come with FindEigen3:

```
william@farl ~/dev/wstool (git:wkentaro-another-wstool-completion:feb5a03)
% ls /usr/local/Cellar/cmake/3.2.3/share/cmake/Modules | grep Eigen

william@farl ~/dev/wstool (git:wkentaro-another-wstool-completion:feb5a03)
% brew info cmake
cmake: stable 3.2.3 (bottled), HEAD
```

We did research on this at one point and I think it is Eigen which is distributing the FindEigen3.cmake file:

https://github.com/ros/cmake_modules/issues/25#issuecomment-63561647

Apparently there was an issue with Eigen's packaging in Homebrew, maybe the OP needs to update the version installed from Homebrew:

https://github.com/ros/cmake_modules/issues/25#issuecomment-68018833

paulbovbel, 2015-08-03:
Thanks @wjwwood. I assumed that anything in `/usr/share/cmake-2.8/Modules` was provided by cmake.

presocratics, 2015-08-13:
It seems this has been resolved, but I'll share how I got this working:
Elsewhere in the ros build workspace I found a copy of FindEigen3.cmake:

    ros_catkin_ws$ cp src/orocos_kinematics_dynamics/orocos_kdl/config/FindEigen3.cmake   src/perception_pcl/pcl_ros/cfg/

Then I removed the following from CMakeLists:

    find_package(Eigen 3 Required)

Replacing it with these lines from the orocos CMakeLists

    find_package(Eigen 3 QUIET)
    if(NOT Eigen_FOUND)
      include(${PROJ_SOURCE_DIR}/cfg/FindEigen3.cmake)
      set(Eigen_INCLUDE_DIR ${EIGEN3_INCLUDE_DIR})
    endif()
    include_directories(${Eigen_INCLUDE_DIR})

To be perfectly honest, I didn't change config -> cfg when I did my build, but it seems to have worked.

bchretien, 2015-08-26:
:+1: `FindEigen3.cmake` is not shipped in Arch's `eigen` package either, however it does provide a pkg-config file which could also be used.

Lome789, 2015-09-27:
@presocratics could you please provide detail about how to remove and replace?

Lome789, 2015-09-28:
i tried all the methods provided and it seems something went wrong...
==> Processing catkin package: 'pcl_ros'
==> Building with env: '/Users/Lome/ros_catkin_ws/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/Users/Lome/ros_catkin_ws/build_isolated/pcl_ros'
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** io features related to openni will be disabled
-- checking for module 'openni2-dev'
--   package 'openni2-dev' not found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** visualization features related to openni will be disabled
-- checking for module 'openni2-dev'
--   package 'openni2-dev' not found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** visualization features related to openni2 will be disabled
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** apps features related to openni will be disabled
-- looking for PCL_COMMON
-- looking for PCL_GEOMETRY
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_SURFACE
-- looking for PCL_REGISTRATION
-- looking for PCL_KEYPOINTS
-- looking for PCL_TRACKING
-- looking for PCL_RECOGNITION
-- looking for PCL_APPS
-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- looking for PCL_CLOUD_COMPOSER
-- looking for PCL_MODELER
-- looking for PCL_POINT_CLOUD_EDITOR
-- looking for PCL_OUTOFCORE
-- looking for PCL_PEOPLE
-- Using CATKIN_DEVEL_PREFIX: /Users/Lome/ros_catkin_ws/devel_isolated/pcl_ros
-- Using CMAKE_PREFIX_PATH: /Users/Lome/ros_catkin_ws/install_isolated
-- This workspace overlays: /Users/Lome/ros_catkin_ws/install_isolated
-- Using PYTHON_EXECUTABLE: /usr/local/bin/python
-- Using default Python package layout
-- Using empy: /Library/Python/2.7/site-packages/em.pyc
[... 99 lines cut ...]
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: ld: library not found for -lvtkRenderingFreeTypeOpenGL
ld: library not found for -lvtkRenderingFreeTypeOpenGL
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
*** [/Users/Lome/ros_catkin_ws/devel_isolated/pcl_ros/lib/pcl_ros/convert_pointcloud_to_image] Error 1
make[2]: *** [/Users/Lome/ros_catkin_ws/devel_isolated/pcl_ros/lib/pcl_ros/convert_pcd_to_image] Error 1
make[2]: *** [/Users/Lome/ros_catkin_ws/devel_isolated/pcl_ros/lib/pcl_ros/pcd_to_pointcloud] Error 1
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/pcd_to_pointcloud.dir/all] Error 2
make[1]: *** [CMakeFiles/convert_pcd_to_image.dir/all] Error 2
[ 26%] Linking CXX executable /Users/Lome/ros_catkin_ws/devel_isolated/pcl_ros/lib/pcl_ros/pointcloud_to_pcd
ld: library not found for -lvtkRenderingFreeTypeOpenGL
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/Lome/ros_catkin_ws/devel_isolated/pcl_ros/lib/pcl_ros/pointcloud_to_pcd] Error 1
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'pcl_ros': 
  Command '['/Users/Lome/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /Users/Lome/ros_catkin_ws/build_isolated/pcl_ros && /Users/Lome/ros_catkin_ws/install_isolated/env.sh make -j4 -l4

Command failed, exiting.

jmtatsch, 2015-10-10:
I guess this should resolve the issue:
ros-perception/perception_pcl/pull/108

timcloss, 2016-02-10:
@bchretien patch worked for me,

slavaGanzin, 2016-03-04:
@bchretien Thanks!

elpetak, 2017-01-12:
Compiling ROS in a Raspberry pi 3 I was able to workaround the Eigen3 error by adding "-DCMAKE_MODULE_PATH=/usr/share/cmake-3.0/Modules" to the catkin_make_isolated command.
It's still running but hopefully that won't cause any other errors.

howiechiang, 2017-01-31:
elpetak, What line do you insert "-DCMAKE_MODULE_PATH=/usr/share/cmake-3.0/Modules" o your raspberry pi 3??

GreatAlexander, 2017-02-05:
@elpetak Thanks, that solved it for me ;)
@howiechiang He meant when you are compiling the code in your workspace, so section 3.3 if you followed the instructions in http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Indigo%20on%20Raspberry%20Pi
For example, the command I use with elpetak's edit is:
`sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release -DCMAKE_MODULE_PATH=/usr/share/cmake-3.0/Modules --install-space /opt/ros/indigo -j3`

soswow, 2017-02-16:
I propose update documentation on http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi peopl

Achllle, 2017-06-12:
The wiki hasn't been updated yet as @soswow stated. It took us a full day of frustration to find this easy working fix by @elpetak and @GreatAlexander.

msy22, 2017-10-23:
Just to add to this for anyone reading later, this issues seems to occur because `FindEigen3.cmake` is included in the version 2.X builds of CMake, but not the version 3.X builds. So normally this file is provided in `usr/share/cmake-X.Y/Modules`, but in cmake versions 3.X this doesn't seem to exist. 

If you have both cmake 2.X and 3.Y versions installed you can solve this problem by copying `FindEigen3.cmake` from one directory to the other, using:

    $  sudo cp /usr/share/cmake-2.8/Modules/FindEigen3.cmake /usr/share/cmake-3.2/Modules/

MadCauchy, 2017-11-21:
@msy22 Thanks a lot! I have the same issue. And your method works!  In my case the command should be 

$ sudo cp /usr/share/cmake-2.8/Modules/FindEigen3.cmake /usr/local/share/cmake-3.2/Modules/

because the cmake-3.2 is under the folder /usr/local/share/ in my ubuntu 14.04.

m782015362, 2019-09-25:
Hello everyone, i am a  novice,when i building the catkin Workspace i got the problem "fatal error: opencv2/xfeatures2d/cuda.hpp: No such file or directory".Do you have any good Suggestions to help me solve it?  thanks

SteveMacenski, 2019-11-19:
Just to recap- are there any action items for this repo to help folks? If not, I can close this and pin it for future searchers

SteveMacenski, 2019-11-26:
Closing - it sounds like things have been resolved.

closed 2019-11-26 by SteveMacenski