ros-perception/perception_pcl #119: get perception_pcl building on kinetic

jgoppert, 2016-04-21:
I got pretty close, but ran into this error:

```bash
Starting  >>> pcl_ros                                                            
_________________________________________________________________________________
Warnings   << pcl_ros:check /home/jgoppert/git/visim/logs/pcl_ros/build.check.006.log
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** visualization features related to openni2 will be disabled
CMake Warning (dev) at CMakeLists.txt:80 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "pcl_ros_generate_messages_cpp" of target
  "pcl_ros_tf" does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/jgoppert/git/visim/build/pcl_ros; catkin build --get-env pcl_ros | catkin env -si  /usr/bin/make cmake_check_build_system; cd -
.................................................................................
_________________________________________________________________________________
Errors     << pcl_ros:make /home/jgoppert/git/visim/logs/pcl_ros/build.make.007.log
/usr/bin/ld: cannot find -lvtkproj4
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jgoppert/git/visim/devel/.private/pcl_ros/lib/pcl_ros/convert_pointcloud_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: cannot find -lvtkproj4
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jgoppert/git/visim/devel/.private/pcl_ros/lib/pcl_ros/pcd_to_pointcloud] Error 1
make[1]: *** [CMakeFiles/pcd_to_pointcloud.dir/all] Error 2
/usr/bin/ld: cannot find -lvtkproj4
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jgoppert/git/visim/devel/.private/pcl_ros/lib/pcl_ros/pointcloud_to_pcd] Error 1
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2

```

jacquelinekay, 2016-04-21:
+1, let's get the pcl pipeline released into Kinetic soon :)

I haven't looked much at the pcl internals before and I don't know much about the history of pcl_ros, but I grepped through the repo and couldn't find another reference to `pcl_ros_generate_messages_cpp`. It looks like a target name that `genmsg` would generate, which doesn't make much sense because `genmsg` isn't referenced directly in `pcl_ros` and message types are provided by `pcl_msgs` and `sensor_msgs`. I think it's safe to remove. I can try testing this with a source build of the pcl packages on Kinetic/Xenial later today.

jgoppert, 2016-04-21:
@jacquelinekay I removed that msg_gen reference and the warning went away, we still have the pcl link error though, I think it is because the cmake config below lists vtk libraries that are not installed on the system: 

```bash
/usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:      set(VTK_LIBRARIES "vtkImagingStencil;vtkCommonComputationalGeometry;vtkCommonDataModel;vtkCommonMath;vtkCommonCore;vtksys;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkImagingCore;vtkCommonExecutionModel;vtkFiltersAMR;vtkFiltersGeneral;vtkFiltersCore;vtkParallelCore;vtkIOLegacy;vtkIOCore;/usr/lib/x86_64-linux-gnu/libz.so;vtkInteractionWidgets;vtkFiltersHybrid;vtkImagingSources;vtkRenderingCore;vtkCommonColor;vtkFiltersExtraction;vtkFiltersStatistics;vtkImagingFourier;vtkalglib;vtkFiltersGeometry;vtkFiltersSources;vtkFiltersModeling;vtkImagingGeneral;vtkImagingHybrid;vtkIOImage;vtkDICOMParser;vtkmetaio;/usr/lib/x86_64-linux-gnu/libjpeg.so;/usr/lib/x86_64-linux-gnu/libpng.so;/usr/lib/x86_64-linux-gnu/libtiff.so;vtkInteractionStyle;vtkRenderingAnnotation;vtkImagingColor;vtkRenderingFreeType;/usr/lib/x86_64-linux-gnu/libfreetype.so;vtkftgl;vtkRenderingVolume;vtkIOParallelNetCDF;vtkParallelMPI;/usr/lib/x86_64-linux-gnu/libnetcdf_c++.so;/usr/lib/x86_64-linux-gnu/libnetcdf.so;/usr/lib/x86_64-linux-gnu/hdf5/serial/lib/libhdf5.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/x86_64-linux-gnu/libsz.so;/usr/lib/x86_64-linux-gnu/libdl.so;/usr/lib/x86_64-linux-gnu/libm.so;/usr/lib/x86_64-linux-gnu/hdf5/serial/lib/libhdf5_hl.so;vtkRenderingOpenGL;vtkIOLSDyna;vtkIOXML;vtkIOGeometry;/usr/lib/x86_64-linux-gnu/libjsoncpp.so;vtkIOXMLParser;/usr/lib/x86_64-linux-gnu/libexpat.so;vtkLocalExample;vtkInfovisCore;vtkGeovisCore;vtkInfovisLayout;vtkViewsCore;vtkproj4;/usr/lib/x86_64-linux-gnu/libpython2.7.so;vtkTestingGenericBridge;/usr/lib/libgl2ps.so;verdict;vtkIOMovie;/usr/lib/x86_64-linux-gnu/libtheoraenc.so;/usr/lib/x86_64-linux-gnu/libtheoradec.so;/usr/lib/x86_64-linux-gnu/libogg.so;vtkFiltersImaging;vtkIOMINC;vtkRenderingLOD;vtkViewsQt;vtkGUISupportQt;vtkViewsInfovis;vtkChartsCore;vtkRenderingContext2D;vtkRenderingLabel;vtkRenderingImage;vtkFiltersFlowPaths;vtkxdmf2;/usr/lib/x86_64-linux-gnu/libxml2.so;vtkFiltersReebGraph;vtkViewsContext2D;vtkIOXdmf2;vtkIOAMR;vtkRenderingContextOpenGL;vtkImagingStatistics;vtkIOParallel;vtkFiltersParallel;vtkIONetCDF;vtkexoIIc;vtkGUISupportQtOpenGL;vtkIOParallelLSDyna;vtkFiltersParallelGeometry;vtkGUISupportQtWebkit;vtkIOPLY;vtkWrappingTools;vtkFiltersHyperTree;vtkRenderingVolumeOpenGL;vtkIOExodus;vtkIOPostgreSQL;vtkIOSQL;sqlite3;vtkWrappingJava;vtkFiltersParallelFlowPaths;vtkFiltersParallelStatistics;vtkFiltersProgrammable;vtkFiltersParallelImaging;vtkRenderingParallelLIC;vtkRenderingLIC;vtkInteractionImage;vtkFiltersPython;vtkWrappingPythonCore;vtkIOParallelExodus;vtkFiltersGeneric;vtkIOVideo;vtkRenderingQt;vtkFiltersTexture;vtkIOInfovis;vtkGUISupportQtSQL;vtkRenderingFreeTypeOpenGL;vtkInfovisBoostGraphAlgorithms;vtkRenderingGL2PS;vtkIOGeoJSON;vtkFiltersVerdict;vtkViewsGeovis;vtkIOImport;vtkTestingIOSQL;vtkPythonInterpreter;vtkIOODBC;vtkIOEnSight;vtkIOMySQL;vtkRenderingMatplotlib;vtkDomainsChemistry;vtkIOExport;vtkFiltersParallelMPI;vtkIOParallelXML;vtkTestingRendering;vtkIOMPIParallel;vtkParallelMPI4Py;vtkFiltersSMP;vtkFiltersSelection;vtkIOVPIC;VPIC;vtkImagingMath;vtkImagingMorphological;vtkRenderingParallel;vtkRenderingFreeTypeFontConfig;vtkIOFFMPEG;vtkIOMPIImage;vtkIOGDAL")
```

jacquelinekay, 2016-04-21:
Yeah--I was just going to post about that. This seems related to the update to libvtk6 in Xenial. Can you check which libvtk you have installed using dpkg?

jgoppert, 2016-04-21:
```bash
dpkg --list | grep vtk
ii  libvtk6-dev                                 6.2.0+dfsg1-10build1                                  amd64        VTK header files
ii  libvtk6-java                                6.2.0+dfsg1-10build1                                  amd64        Visualization Toolkit - A high level 3D visualization library - java
ii  libvtk6-qt-dev                              6.2.0+dfsg1-10build1                                  amd64        VTK header files, containing Qt files
ii  libvtk6.2                                   6.2.0+dfsg1-10build1                                  amd64        VTK libraries
ii  libvtk6.2-qt                                6.2.0+dfsg1-10build1                                  amd64        VTK libraries, Qt files
ii  python-vtk6                                 6.2.0+dfsg1-10build1                                  amd64        Python bindings for VTK
ii  tcl-vtk6                                    6.2.0+dfsg1-10build1                                  amd64        Tcl bindings for VTK
ii  vtk6                                        6.2.0+dfsg1-10build1                                  amd64        Binaries for VTK6
```

jgoppert, 2016-04-21:
All of the other ros packages are built against vtk6. Maybe pcl was built against vtk5 by accident? Or maybe just a manual library link hack that needs to be updated.

jacquelinekay, 2016-04-21:
yeah, it appears to me that pcl has a dependency on vtk6. So this does seem like a problem with how pcl was released into Ubuntu.

http://packages.ubuntu.com/source/xenial/pcl

jacquelinekay, 2016-04-21:
Interesting, I get a slightly different error message from you:

```
Errors     << pcl_ros:make /home/jackie/pcl_ws/logs/pcl_ros/build.make.000.log                                                                                     
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj.so', needed by '/home/jackie/pcl_ws/devel/.private/pcl_ros/lib/pcl_ros/pointcloud_to_pcd'.  Stop.
make[2]: *** Waiting for unfinished jobs....
```

it appears that the path `/usr/lib/x86_64-linux-gnu/libproj.so` is hardcoded into `VTKTargets.cmake`, but the library is instead called `libproj.so.9` is installed when libvtk6 is installed.

I think your workspace and my workspace are resolving different branches of `PCLConfig.cmake`.

I am going to make some minimal examples to reproduce these errors that use pure CMake so that I can report them to the maintainers of pcl and vtk (if it's necessary).

jgoppert, 2016-04-21:
Weird, this is a fresh system started from the xenial beta2 iso. But after doing

```
sudo apt-get install libpcl-dev --reinstall
```

I have a new cmake file that doesn't have the proj4 reference.

I already started bugging the maintainer here: https://bugs.launchpad.net/ubuntu/+source/pcl/+bug/1573174

Will close if I can manage to get it working with new files after doing a complete reinstall of pcl.

jacquelinekay, 2016-04-21:
yep. A workaround would be to add a dependency on libproj-dev in the package.xml. But really I think libvtk should fix their dependencies to include libproj-dev instead of libproj-9.

After I install libproj-dev I get the same error as you, even for a minimal Cmake example.

jgoppert, 2016-04-21:
Building pcl 1.8 from source works for me without modification, here is a deb created with checkinstall, unfortunately cpack -G DEB gave an error:
https://drive.google.com/open?id=0B3wIEs1gcVahbUJ5aFE3Zmo1YjQ

So is it possible that this is just pcl 1.7 and vtk6 related?

jacquelinekay, 2016-04-21:
that's good to know that pcl 1.8 works from source.

The pcl maintainer said that an upstream fix is needed in vtk, apparently it's been fixed but the new version isn't in Ubuntu yet.

https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1573234

I've been playing around with a hack to take vtkproj4 out of the link dependencies in the pcl_ros CMakeLists but I haven't gotten it working.

jgoppert, 2016-04-21:
@jacquelinekay Try this PR: https://github.com/ros-perception/perception_pcl/pull/120

jacquelinekay, 2016-04-21:
nice. I just opened the last approach I was talking about (#121) which makes the source build work with the libpcl1.7 deb but is also quite a hack. I guess it's up to the maintainers to decide which approach to take. I don't really care which one, as long as we get this repo into Kinetic soon. cheers!

jgoppert, 2016-04-21:
@jacquelinekay Thanks for the help today. I will try your PR, obviously much better than building pcl from source. Glad we got it working! :-)

jacquelinekay, 2016-04-21:
I'm getting a link error in `bag_to_pcd` now:

```
CMakeFiles/bag_to_pcd.dir/tools/bag_to_pcd.cpp.o: In function `bool rosbag::MessageInstance::isType<tf::tfMessage_<std::allocator<void> > >() const':
/home/jackie/pcl_ws/src/perception_pcl/pcl_ros/tools/bag_to_pcd.cpp:(.text._ZNK6rosbag15MessageInstance6isTypeIN2tf10tfMessage_ISaIvEEEEEbv[_ZNK6rosbag15MessageInstance6isTypeIN2tf10tfMessage_ISaIvEEEEEbv]+0x81): undefined reference to `rosbag::MessageInstance::getMD5Sum() const'
CMakeFiles/bag_to_pcd.dir/tools/bag_to_pcd.cpp.o: In function `bool rosbag::MessageInstance::isType<sensor_msgs::PointCloud2_<std::allocator<void> > >() const':
/home/jackie/pcl_ws/src/perception_pcl/pcl_ros/tools/bag_to_pcd.cpp:(.text._ZNK6rosbag15MessageInstance6isTypeIN11sensor_msgs12PointCloud2_ISaIvEEEEEbv[_ZNK6rosbag15MessageInstance6isTypeIN11sensor_msgs12PointCloud2_ISaIvEEEEEbv]+0x81): undefined reference to `rosbag::MessageInstance::getMD5Sum() const'
```

paulbovbel, 2016-04-22:
Thanks for the fix @jacquelinekay, I've tested and get a clean build with your PR

I'll have to get caught up on the issue with the vtk version change and try to figure out a long-term solution

jolting, 2016-05-03:
The bug is already fixed in Yakkety. We just need it in Xenial. Everyone click that link that says this bug affects you.

https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1573234

tfoote, 2016-05-06:
Although this now builds it's still propogating to downstream packages ala: https://github.com/ros-planning/navigation/issues/456 It would be great to either add the dependency until the ubuntu ticket is closed or find a way to remove the linking requirement for downstream packages.

jacquelinekay, 2016-05-17:
As mentioned by Tully in a few other issues, he successfully set up a build of vtk with a fix for the missing vtkproj issue. @paulbovbel feel free to revert my PR as it shouldn't be needed anymore.

flixr, 2016-08-02:
Any info on getting perception_pcl for xenial/kinetic on armhf?

jolting, 2016-08-04:
@flixr A while back, I was having a problem with building the pcl library on armhf when I backported some fixes to xenial. I'm not sure if that's still an issue. It looked like some missing OpenGL includes.
I'm not sure if it's been fixed and packaged, but if PCL won't build than perception_pcl won't either.

References:
https://launchpad.net/~jolting/+archive/ubuntu/backport/+packages
https://launchpadlibrarian.net/258687214/buildlog_ubuntu-xenial-armhf.pcl_1.7.2-14ubuntu4~ubuntu16.04.1~ppa1_BUILDING.txt.gz

tfoote, 2016-08-10:
We're blocked on libpcl-dev being missing. This ticket is tracking that: https://github.com/ros/rosdistro/issues/11583

I haven't had a chance to try building pcl in my ppa as suggested by @jspricke. I was hoping that vtk upstream might have been fixed and enabled the upstream libpcl-dev build: https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1573234

ojura, 2017-06-23:
Hey guys, a related issue is causing build problems on Zesty. `VTK_LIBRARIES` specifies `libmpi.so`, and this breaks linking anything with libpcl. I tried reaching out at various places, including at the PCL site, but haven't received a response. Here's the launchpad bug:

https://bugs.launchpad.net/ubuntu/+source/pcl/+bug/1694397

jolting, 2017-06-23:
I think this bug was reported a while ago.
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=848793
However, I don't think it's fixed.

If you want a quick workaround use install `libmpich-dev`. I wouldn't recommend this as a permanent solution.

ojura, 2017-06-23:
The way I fixed this is by copying this repository's mitigation:
https://github.com/ros-perception/perception_pcl/blob/b53b5d5c3abc7c3afde8a77177674e6d74c59522/pcl_ros/CMakeLists.txt#L12

ndwong, 2017-06-28:
Hello, I've been trying to fix this problem as well. In my case, I'm building it for Kinetic on Ubuntu 16 running off an nVidia Jetson TX2 (arm64). 

I can't install pcl_ros off the repo due to VTK problems,  and compiling from source led me here. 
Tried permutations of removing  "vtkproj4" and "/usr/lib/libmpi.so" from the cmake, to no avail. 

Anyone know of something else that I can try?

ojura, 2017-06-28:
Do you have a missing target error (which is raised by make), or a linking error? The original issue was for a linking error. I kind of hijacked it with my comment, as it is not the completely same bug, although it's related.

developerAy, 2018-07-10:
Hello. I was trying to customize dwa-local-planner & base-local-planner in the ROS navigation stack. In order to do that, I had to use catkin_make to compile it in the catkin workspace. I got the same error using kinetic ROS which is:

 make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj4.so', needed by '/home/username/catkin_ws/devel/lib/pcl_ros/lib/libdwa-local-planner.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....

What I did to solve this problem was using this tricky command which is not recommended as a fundamental solution (But solves the problem for now): 

sudo ln -s /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so /usr/lib/x86_64-linux-gnu/libproj4.so