ros-perception/perception_pcl #130: Lunar Release

mikaelarguedas, 2017-04-07:
Hi @paulbovbel and `perception_pcl` maintainers!

As mentioned in ros-perception/pcl_msgs#7, it would be great to have this repo released into Lunar asap to unblock the release of the navigation stack and all downstream packages. Please run bloom release on the current 1.4.1 version to release it into Lunar.

Thanks!

paulbovbel, 2017-04-11:
Looks like PCL 1.8 in yakkety/zesty has split into multiple packages (http://packages.ubuntu.com/search?keywords=libpcl), where there is no longer a debian like `libpcl1.8`.

Will have to investigate how CMakeLists and rosdep dependencies will change in lunar, given that we can no longer just `find_package(PCL 1.7 REQUIRED)`.

mikaelarguedas, 2017-04-11:
hmm that's unconvenient. Thanks for looking into this!
If it's of any help, the `ros_buildfarm` prerelease jobs can help you test this on the new platforms (setting up containers and building the code). More info about running prerelease for repositories not part of the rosdistro index can be found [here](https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/jobs/prerelease_jobs.rst#run-for-custom-repositories)

wkentaro, 2017-04-13:
> Looks like PCL 1.8 in yakkety/zesty has split into multiple packages (http://packages.ubuntu.com/search?keywords=libpcl), where there is no longer a single debian like libpcl1.7.

I'm not sure, but `libpcl-dev` not work? Adding it to after [this line](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2196)?

mikaelarguedas, 2017-04-13:
I think that @paulbovbel is referring to how to find the package via CMake code and not the global rosdep key. The package `libpcl-dev` contains the package libpcl1.7 [on xenial](http://packages.ubuntu.com/xenial/libpcl-dev) so you can `find_package(PCL 1.7 REQUIRED)`.
But in [yakkety and later](http://packages.ubuntu.com/yakkety/libpcl-dev) there is no package libpcl1.8 but multiple libpclFEATURE1.8 so you may need to find_package them independently.

I didnt try to run `find_package(PCL 1.8 REQUIRED)` though...

wkentaro, 2017-04-13:
I see.

BTW, are there any tools to test this kind of thing? Do you use docker + rosdep + wstool + rosinstall_generator?

wkentaro, 2017-04-13:
**Xenial**

OK

**Yakkety**

OK with https://github.com/ros/rosdistro/pull/14524

**Zesty**

mikaelarguedas, 2017-04-14:
@wkentaro it looks like you figured out ways to test it. I'm answering here anyway because that can be useful for future releases. Usually I test things using the prerelease scripts provided by `python3-ros-buildfarm`.
If your package has a source entry on rosdistro you can use [prerelease.ros.org](http://prerelease.ros.org/)
Otherwise you can use the prerelease scripts on custom repositories (not indexed in rosdistro) folowing these instructions https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/jobs/prerelease_jobs.rst#run-for-custom-repositories

wkentaro, 2017-04-15:
Oh, this should be what I was looking for. Thanks!

paulbovbel, 2017-04-25:
Please branch lunar-devel, and remove pointcloud_to_laserscan from it before blooming. That package has been here (https://github.com/ros-perception/pointcloud_to_laserscan) for some time.

paulbovbel, 2017-04-25:
Also don't forget to add yourself as a maintainer to all the packages :) otherwise you won't get emails or karma

wkentaro, 2017-04-26:
Sent PR to ros/rosdistro: https://github.com/ros/rosdistro/pull/14672

mikaelarguedas, 2017-04-26:
The initial builds were failing because of a wrong rosdep key (https://github.com/ros/rosdistro/pull/14676)
Now that https://github.com/ros/rosdistro/pull/14676 has been merged, can you please rerun bloom to trigger packages with the right key?
Thanks!

wkentaro, 2017-04-26:
https://github.com/ros/rosdistro/pull/14677

mikaelarguedas, 2017-04-26:
Looks like all jobs turned green, I'm going to close this. Thanks @wkentaro for the release!

closed 2017-04-26 by mikaelarguedas

paulbovbel, 2017-04-26:
Thanks @wkentaro!