ros-perception/perception_pcl #139: [pcl_ros] Qt5::Widgets not found on Debian stretch

mikaelarguedas, 2017-04-27:
pcl_ros binary jobs fail at linktime: `/usr/bin/ld: cannot find -lQt5::Widgets` 
Job link: http://build.ros.org/view/Lbin_ds_dS64/job/Lbin_ds_dS64__pcl_ros__debian_stretch_amd64__binary/2/console
timestamp:"13:17:33"

wkentaro, 2017-04-27:
I've tried to fix this, but still not sure how to install Qt5::Widget library.

paulbovbel, 2017-04-27:
maybe https://packages.debian.org/sid/libqwt-qt5-6?

the real question is why that fails only for debian stretch....

wkentaro, 2017-04-28:
libqt5widgets5 is installed, but it still fails

```
......................................................................................................
______________________________________________________________________________________________________
Errors     << pcl_ros:make /root/catkin_ws/logs/pcl_ros/build.make.000.log
/usr/bin/ld: cannot find -lQt5::Widgets
collect2: error: ld returned 1 exit status
make[2]: *** [/root/catkin_ws/devel/.private/pcl_ros/lib/pcl_ros/convert_pcd_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pcd_to_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: cannot find -lQt5::Widgets
collect2: error: ld returned 1 exit status
make[2]: *** [/root/catkin_ws/devel/.private/pcl_ros/lib/pcl_ros/convert_pointcloud_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
make: *** [all] Error 2
cd /root/catkin_ws/build/pcl_ros; catkin build --get-env pcl_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................................................................
Failed     << pcl_ros:make                         [ Exited with code 2 ]
Failed    <<< pcl_ros                              [ 1 minute and 23.6 seconds ]
[build] Summary: 1 of 2 packages succeeded.
[build]   Ignored:   1 packages were skipped or are blacklisted.
[build]   Warnings:  1 packages succeeded with warnings.
[build]   Abandoned: None.
[build]   Failed:    1 packages failed.
[build] Runtime: 1 minute and 33.6 seconds total.
[build] Note: Workspace packages have changed, please re-source setup files to use them.
root@1827d2d78f77:~/catkin_ws# sudo apt install libqt5widgets5
Reading package lists... Done
Building dependency tree
Reading state information... Done
libqt5widgets5 is already the newest version (5.7.1+dfsg-3+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

paulbovbel, 2017-04-28:
Check out https://github.com/googlecartographer/cartographer_ros/issues/119

wkentaro, 2017-04-29:
Actually it still fails. 
http://build.ros.org/view/Lbin_ds_dS64/job/Lbin_ds_dS64__pcl_ros__debian_stretch_amd64__binary/8/console

wkentaro, 2017-05-10:
Fixed via https://github.com/ros-perception/perception_pcl/pull/141

PR #140 (2017-04-28): Find Qt5Widgets to fix -lQt5::Widgets error
Fix #139
closed 2017-05-10 by wkentaro