ros-perception/perception_pcl #152: PointCloud2 should have compressed transport option

rconnorlawson, 2017-07-23:
Similar to [image_transport](http://wiki.ros.org/image_transport) and [compressed_image_transport](http://wiki.ros.org/compressed_image_transport), there should be support for compressed transmission of pointclouds in ROS. PCL has built-in support for real-time compression of pointcloud streams ([paper](http://ieeexplore.ieee.org/document/6224647/), [tutorial](http://pointclouds.org/documentation/tutorials/compression.php), and [documentation](http://docs.pointclouds.org/trunk/classpcl_1_1io_1_1_octree_point_cloud_compression.html)), so adding this feature should be straightforward.

Even for moderately sized RGB pointclouds (such as those generated by an Intel RealSense or Kinect), the bandwidth required to transfer the clouds at 30 Hz exceeds 1Gbps. Neglecting overhead, 640 * 480 pixels * 32 bpp * 8 bits * 30 Hz = 2.19 Gbps. Almost no one has a 10Gbps connection available on their robot; compression is required to achieve distributed use of pointclouds at camera rate on most networks.

Along with improvements in transmission rate, a plugin structure like that provided by image_transport would also support further research into pointcloud compression techniques.

peci1, 2018-11-21:
https://answers.ros.org/question/298352/new-message-format-for-compressed-pointcloud2/

paplhjak, 2019-08-12:
I have implemented a [point_cloud_transport](https://github.com/paplhjak/point_cloud_transport) package for transport of PointCloud2 messages, which uses plugin interface ala image_transport. In the README, you can also find links to other repositories, which contain but are not limited to:

1. plugin for compression using Google Draco
2. template for implementing your custom plugins
3. tutorial on how to use the package

Any help in further development of the project is welcome and appreciated.

Hope that this helps.