ros-perception/perception_pcl #166: Build fail on yakkety

wkentaro, 2017-11-15:
https://travis-ci.org/ros-perception/perception_pcl/jobs/283075014
```
[31m[1mCMake Error[0m at /home/travis/catkin_ws/src/perception_pcl/pcl_ros/CMakeLists.txt:5 (find_package):
  By not providing "Findcmake_modules.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "cmake_modules", but CMake did not find one.

  Could not find a package configuration file provided by "cmake_modules"
  with any of the following names:

    cmake_modulesConfig.cmake
    cmake_modules-config.cmake

  Add the installation prefix of "cmake_modules" to CMAKE_PREFIX_PATH or set
  "cmake_modules_DIR" to a directory containing one of the above files.  If
  "cmake_modules" provides a separate development package or SDK, be sure it
  has been installed.
```

wkentaro, 2017-11-15:
@mikaelarguedas Could you please help me a little to solve this?
I'm testing on `yakkety` with docker. And found that there is apt key, so I can do `apt-get install ros-lunar-cmake-modules` but the apt key can not be resolved by rosdep.
Maybe `yakkety` is removed from release targets of lunar? https://github.com/ros/rosdistro/blob/master/lunar/distribution.yaml#L11-L12

```
root@917d4b78ab30:/# apt search ros-lunar-cmake-modules
Sorting... Done
Full Text Search... Done
ros-lunar-cmake-modules/yakkety 0.4.1-0yakkety-20170801-113830-0800 amd64
  A common repository for CMake Modules which are not distributed with CMake but are commonly used by ROS packages.

root@917d4b78ab30:/# rosdep resolve cmake_modules

ERROR: No definition of [cmake_modules] for OS version [yakkety]

No definition of [cmake_modules] for OS version [yakkety]
	rosdep key : cmake_modules
	OS name    : ubuntu
	OS version : yakkety
	Data: _is_ros: true
debian:
  stretch:
    apt:
      packages:
      - ros-lunar-cmake-modules
fedora:
  '26':
    yum:
      packages:
      - ros-lunar-cmake-modules
osx:
  homebrew:
    packages:
    - ros/lunar/cmake_modules
ubuntu:
  xenial:
    apt:
      packages:
      - ros-lunar-cmake-modules
  zesty:
    apt:
      packages:
      - ros-lunar-cmake-modules
```

paulbovbel, 2017-11-15:
That's definitely the case, yakkety went EOL in July - guess we should remove it from the travis tests?

mikaelarguedas, 2017-11-15:
> Maybe yakkety is removed from release targets of lunar? 

Yes yakkety was removed from the lunar targets in August (https://discourse.ros.org/t/suspension-of-debian-packaging-for-eol-ubuntu-yakkety/2444). We will keep hosting the deb packages but will not be running any jobs for it anymore and tools like bloom and rosdep are not aware of it anymore as it was removed from the distibution file (https://github.com/ros/rosdistro/pull/15898/files).

wkentaro, 2017-11-15:
Thanks! BTW, where can I find that kind of information? I could not find that information at roswiki: http://wiki.ros.org/Distributions.

mikaelarguedas, 2017-11-15:
Our current policy is that we stop iterating/buiding packages for Ubuntu distributions right after they become EOL. As we keep hosting the binaries, the distributions are still "available" (you can download and install the last packages built before dropping support), we dont remove them from the wiki.
Every time we will stop building for a given ROS Distribution or OS we send an announcement on Discourse and a few weeks later remove them from the distribution file.

wkentaro, 2017-11-15:
I see. Thanks! I will try to keep checking discourse next time.

closed 2017-11-15 by wkentaro