ros-perception/perception_pcl #170: Release the latest version on indigo?

furushchev, 2018-01-13:
Since the latest version of `pcl_ros` package is now `1.5.3` and many fixes and improvements are included, it's time to port this into `indigo` and other distro.
(Note that the version of this package on indigo is still `1.2.7` )

wkentaro, 2018-03-31:
If you would like the use the latest commits in `lunar-devel`, first you need to cherry-pick commits from `lunar-devel` to `indigo-devel`.
https://github.com/ros-perception/perception_pcl/commits/indigo-devel

wkentaro, 2018-03-31:
FYI, `1.2` of `1.2.7` means for `indigo`
1.2: indigo
1.3: jade
1.4: kinetic
1.5: lunar
so 1.5.3 is only for `lunar`.

furushchev, 2018-03-31:
@wkentaro Thanks for clear explanation.
I see. Is there still any chance to port only bugfixes?

Nodelet declaration in `pcl_ros` is currently malformed and many nodes does not work on indigo.
https://github.com/ros-perception/perception_pcl/blob/1.2.7/pcl_ros/pcl_nodelets.xml

wkentaro, 2018-03-31:
I think changes that have the backcompatibility can be backported.

furushchev, 2018-03-31:
It will be nice if these are ported into older distros:
https://github.com/ros-perception/perception_pcl/pull/134
https://github.com/ros-perception/perception_pcl/pull/165
https://github.com/ros-perception/perception_pcl/pull/153
https://github.com/ros-perception/perception_pcl/pull/160
https://github.com/ros-perception/perception_pcl/pull/147

(I may overlook other necessary pull requests)

wkentaro, 2018-03-31:
Sorry, I found backporting all of them is a bit difficult (because they are not backported even for jade).
So I ended up backporting the few commits here: https://github.com/ros-perception/perception_pcl/pull/180
Could you check it?

wkentaro, 2018-03-31:
The kinetic will have almost all changes that are included in lunar: https://github.com/ros-perception/perception_pcl/pull/178

furushchev, 2018-03-31:
I leave a comment here since I don't have the right to review your pull request.
Basically looks good to me, it would be better if we got test codes backported so to check if future changes on these distros break the functionality.
Anyway, thank you very much!