ros-perception/perception_pcl #174: Unable to compile with Kinetic on Arm64

Russ76, 2018-03-28:
On a Jetson TX1 with Xenial and Kinetic, compiler sticks at 26%. The first time I did a hard "turn off." That's why some was finished in the output below.

The second time, after 10 minutes stuck, it threw this error:

nvidia@tegra-ubuntu:~/perception_ws$ catkin_make
Base path: /home/nvidia/perception_ws
Source space: /home/nvidia/perception_ws/src
Build space: /home/nvidia/perception_ws/build
Devel space: /home/nvidia/perception_ws/devel
Install space: /home/nvidia/perception_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nvidia/perception_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/nvidia/perception_ws/build"
####
[  0%] Built target bond_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target pcl_msgs_generate_messages_eus
[  0%] Built target nodelet_topic_tools_gencfg
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target pcl_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target bond_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target bond_generate_messages_py
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target nodelet_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target pcl_msgs_generate_messages_py
[  0%] Built target bond_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target nodelet_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target nodelet_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target pcl_msgs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_nodejs
[  0%] Built target bond_generate_messages_lisp
[  0%] Built target nodelet_generate_messages_eus
[  0%] Built target topic_tools_generate_messages_cpp
[  0%] Built target topic_tools_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target pcl_msgs_generate_messages_cpp
[  0%] Built target std_srvs_generate_messages_eus
[  0%] Built target tf_generate_messages_cpp
[  0%] Built target topic_tools_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target topic_tools_generate_messages_nodejs
[  0%] Built target topic_tools_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target tf_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target actionlib_generate_messages_eus
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target nodelet_generate_messages_py
[ 20%] Built target pcl_ros_gencfg
[ 20%] Built target roscpp_generate_messages_cpp
[ 23%] Building CXX object perception_pcl/pcl_ros/CMakeFiles/convert_pcd_to_image.dir/tools/convert_pcd_to_image.cpp.o
[ 23%] Building CXX object perception_pcl/pcl_ros/CMakeFiles/convert_pointcloud_to_image.dir/tools/convert_pointcloud_to_image.cpp.o
[ 25%] Building CXX object perception_pcl/pcl_ros/CMakeFiles/pcd_to_pointcloud.dir/tools/pcd_to_pointcloud.cpp.o
[ 26%] Building CXX object perception_pcl/pcl_ros/CMakeFiles/pointcloud_to_pcd.dir/tools/pointcloud_to_pcd.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
perception_pcl/pcl_ros/CMakeFiles/pointcloud_to_pcd.dir/build.make:62: recipe for target 'perception_pcl/pcl_ros/CMakeFiles/pointcloud_to_pcd.dir/tools/pointcloud_to_pcd.cpp.o' failed
make[2]: *** [perception_pcl/pcl_ros/CMakeFiles/pointcloud_to_pcd.dir/tools/pointcloud_to_pcd.cpp.o] Error 4
CMakeFiles/Makefile2:4028: recipe for target 'perception_pcl/pcl_ros/CMakeFiles/pointcloud_to_pcd.dir/all' failed
make[1]: *** [perception_pcl/pcl_ros/CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 30%] Linking CXX executable /home/nvidia/perception_ws/devel/lib/pcl_ros/convert_pcd_to_image
[ 30%] Linking CXX executable /home/nvidia/perception_ws/devel/lib/pcl_ros/pcd_to_pointcloud
[ 31%] Linking CXX executable /home/nvidia/perception_ws/devel/lib/pcl_ros/convert_pointcloud_to_image
[ 31%] Built target pcd_to_pointcloud
[ 31%] Built target convert_pcd_to_image
[ 31%] Built target convert_pointcloud_to_image
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

k-okada, 2018-03-29:
did you try `catkin_make -j1 -l1` ?

Russ76, 2018-03-29:
Excellent! It was slow but it worked. Where is the reference sheet to learn these commands?

k-okada, 2018-03-30:
`catkin_make --help` will tell you more about  command options.
```
$ catkin_make --help
usage: catkin_make [-h] [-C DIRECTORY] [--source SOURCE] [--build BUILD]
                   [--use-ninja] [--use-nmake] [--force-cmake] [--no-color]
                   [--pkg PKG [PKG ...]]
                   [--only-pkg-with-deps ONLY_PKG_WITH_DEPS [ONLY_PKG_WITH_DEPS ...]]
                   [--cmake-args [CMAKE_ARGS [CMAKE_ARGS ...]]]
                   [--make-args [MAKE_ARGS [MAKE_ARGS ...]]]
                   [--override-build-tool-check]

Creates the catkin workspace layout and invokes cmake and make. Any argument
starting with "-D" will be passed to the "cmake" invocation. The -j (--jobs)
and -l (--load-average) arguments for make are also extracted and passed to
make directly. If no -j/-l arguments are given, then the MAKEFLAGS environment
variable is searched for -j/-l flags. If found then no -j/-l flags are passed
to make explicitly (as not to override the MAKEFLAGS). If MAKEFLAGS is not set
then the job flags in the ROS_PARALLEL_JOBS environment variable are passed to
make. Note: ROS_PARALLEL_JOBS should contain the exact job flags, not just a
number. See:
http://www.ros.org/wiki/ROS/EnvironmentVariables#ROS_PARALLEL_JOBS If
ROS_PARALLEL_JOBS is not set then the flags "-jn -ln" are used, where n is
number of CPU cores. If the number of CPU cores cannot be determined then no
flags are given to make. All other arguments (i.e. target names) are passed to
the "make" invocation. To ignore certain packages place a file named
CATKIN_IGNORE in the package folder. Or you can pass the list of package names
to the CMake variable CATKIN_BLACKLIST_PACKAGES. For example: catkin_make
-DCATKIN_BLACKLIST_PACKAGES="foo;bar".

optional arguments:
  -h, --help            show this help message and exit
  -C DIRECTORY, --directory DIRECTORY
                        The base path of the workspace (default '.')
  --source SOURCE       The path to the source space (default
                        'workspace_base/src')
  --build BUILD         The path to the build space (default
                        'workspace_base/build')
  --use-ninja           Use 'ninja' instead of 'make'
  --use-nmake           Use 'nmake' instead of 'make'
  --force-cmake         Invoke 'cmake' even if it has been executed before
  --no-color            Disables colored output (only for catkin_make and
                        CMake)
  --pkg PKG [PKG ...]   Invoke 'make' on specific packages only
  --only-pkg-with-deps ONLY_PKG_WITH_DEPS [ONLY_PKG_WITH_DEPS ...]
                        Whitelist only the specified packages and their
                        dependencies by setting the CATKIN_WHITELIST_PACKAGES
                        variable. This variable is stored in CMakeCache.txt
                        and will persist between CMake calls unless explicitly
                        cleared; e.g. catkin_make
                        -DCATKIN_WHITELIST_PACKAGES="".
  --cmake-args [CMAKE_ARGS [CMAKE_ARGS ...]]
                        Arbitrary arguments which are passes to CMake. It must
                        be passed after other arguments since it collects all
                        following options.
  --make-args [MAKE_ARGS [MAKE_ARGS ...]]
                        Arbitrary arguments which are passes to make. It must
                        be passed after other arguments since it collects all
                        following options. This is only necessary in
                        combination with --cmake-args since else all unknown
                        arguments are passed to make anyway.
  --override-build-tool-check
                        use to override failure due to using differnt build
                        tools on the same workspace.
```

closed 2018-04-24 by wkentaro