ros-perception/perception_pcl #179: Release 1.4.2 to kinetic

wkentaro, 2018-03-31:


k-okada, 2018-04-08:
This cause regression on downstream packages (http://build.ros.org/view/Kbin_uX64/job/Kbin_uxv8_uXv8__pcl_ros__ubuntu_xenial_arm64__binary/) due to missing `libpcl_visualizatoin` in `pcl_ros_LIBRARIES`
see https://github.com/jsk-ros-pkg/jsk_recognition/pull/2261#issue-180145927 for detail

wkentaro, 2018-04-08:
Do you have any rule declaration for this? ie. REP
I think it was included because of the laziness of the programmer who added that line.

In python community, this kind of implicit dependency is not recommended (just I think actually I couldn't find PEP for this),
because people prefer *explicit* rather than *implicit*.
ex. `boto3`: python client for aws console management: `botocore` is declared by `boto3` even though it is already included in `s3transfer`

```
boto3==1.6.11
  - botocore [required: <1.10.0,>=1.9.11, installed: 1.9.11]
    - docutils [required: >=0.10, installed: 0.14]
    - jmespath [required: <1.0.0,>=0.7.1, installed: 0.9.3]
    - python-dateutil [required: <2.7.0,>=2.1, installed: 2.7.2]
      - six [required: >=1.5, installed: 1.11.0]
  - jmespath [required: <1.0.0,>=0.7.1, installed: 0.9.3]
  - s3transfer [required: <0.2.0,>=0.1.10, installed: 0.1.13]
    - botocore [required: <2.0.0,>=1.3.0, installed: 1.9.11]
      - docutils [required: >=0.10, installed: 0.14]
      - jmespath [required: <1.0.0,>=0.7.1, installed: 0.9.3]
      - python-dateutil [required: <2.7.0,>=2.1, installed: 2.7.2]
        - six [required: >=1.5, installed: 1.11.0]
    - futures [required: <4.0.0,>=2.2.0, installed: 3.2.0]
```

wkentaro, 2018-04-08:
@paulbovbel @mikeferguson Do you have some information or opinion for this?

k-okada, 2018-04-08:
from ROS programmer's point of view, will should depend on ros_pcl not PCL, just like you depend on cv_bridge not OpenCV version itself (http://wiki.ros.org/indigo/Migration#OpenCV).
Since ROS package supports multiple platforms/distributions so that sometimes directly depends on system library will cause trouble and pcl_ros/cv_bridge will gives you interface between ROS package and system library.

wkentaro, 2018-04-08:
I know the case of `cv_bridge` for `OpenCV` but didn't know the that of `pcl_ros`. (also it is only for rosdep key, right?)
I think `cv_bridge` promised that in the migration wiki page for indigo, but can't find any declaration for `pcl_ros`.

wkentaro, 2018-04-08:
It seems `cv_bridge` also does not provide all OpenCV libraries:
https://github.com/ros-perception/vision_opencv/blob/9d49971aa9b32a662a91d9240d8a8b46dc60ca90/cv_bridge/CMakeLists.txt#L16-L22

paulbovbel, 2018-04-08:
I would agree with @wkentaro - pcl_ros should link against what it uses internally. IMO it's not this packages job to inject all of PCL into catkin_LIBRARIES, amd at the very least we want to avoid introducing Gui depends dependencies into a package that's often used on headless machines.

I believe it is this package's job to provide documentation to ROS users how to integrate with PCL, so we should probably use this case to improve documentation.

paulbovbel, 2018-04-08:
However, a backport should not break dependencies, especially in an LTS release. Would you mind trimming this backport, and then we can focus on improving this story for lunar/melodic?

wkentaro, 2018-04-08:
I released 1.4.3: https://github.com/ros/rosdistro/pull/17382 with the fix
for this issue.

k-okada, 2018-04-08:
thanks for quick release.

> It seems cv_bridge also does not provide all OpenCV libraries for Kinetic:

Year between 1.12.4 and 1.12.5....without any pull request.....

Depending on internal use and do not want to GUI library makes sense, And recent cmake prefer to have explicit dependency, but at the same time I still prefer the way `catkin` helped lazy people so far. 

For example, If you want to link exact components, `pcl_ros` can use something like `PCL_IO_LIBRARIES` and dependent package can use `target_link_library(hoge pcl_ros)`  and this is what normal features that CMake provies. Using catkin offers another way of linking libraries using `catkin_LIBRARIES`. So if you go to normal CMake way, i.e. writing minimum set of libraries explicitly,  I think we do not need to use `catkin_LIBRARIES`. On the other hand, If you make `catkin` package, we'd follow catkin way, and I suppose it is designed for what we can not achieve with normal CMake features and may be that is for  lazy people,

Ironically `ament` seems much more CMake way , and they do not offer `ament_LIBRARY` 
https://gist.github.com/dirk-thomas/83ae6bfbfc94e06cd519
but also provide `ament_auto` for lazy people
https://gist.github.com/dirk-thomas/a76f952d05e7b21b0128

Of course, it is maintainer's decision how to do and agree with it if there is explicit documentation for that.