ros-perception/perception_pcl #186: Wiki shows wrong time assignment

PPokorski, 2018-04-24:
Hi! 
I stumbled across an error at [the wiki page](http://wiki.ros.org/pcl_ros#CA-bcc883441139b09a09cf15205453910ae46b3635_21). At the given line the time is taken as `ros::Time::now::toNSec()` , whereas PCL treats the stamp as microseconds, which is even noted [here](https://github.com/ros-perception/pcl_conversions/blob/indigo-devel/include/pcl_conversions/pcl_conversions.h#L79-L89). I ask for updating the wiki, as this error is really not obvious to track down.

wkentaro, 2018-04-24:
This is the different time treatments between ROS and PCL,
and the instruction in the Wiki page is correct.

I am sure that this difference causes some synchronization problems like https://github.com/ros-perception/perception_pcl/issues/115, but this does not mean we should use microseconds in ROS.

PPokorski, 2018-04-24:
Actually if it didn't cause a problem for me, I wouldn't add this issue. Let's consider an example:
Current ros walltime is 100 seconds, 0 nanoseconds.
ros::Time::now()::toNSecs() returns 100,000,000,000
And this value is taken as stamp of **pcl::PointCloud<pcl::PointXYZ>** (note that the msg variable is actually this type). As stamp of PCL PointCloud is interpreted as **microseconds** since epoch, after publishing this pointcloud the stamp will be:
0 seoncds, 100,000,000 nanoseconds.

wkentaro, 2018-04-24:
Oh, sorry. I overlooked that it uses `pcl::PointCloud` instead of `sensor_msgs::PointCloud2`
which I always use to publish in ROS.

Yeah, as you mentioned. It is a bug in the tutorial, and it raises below error:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
```

and the sample code should be

```c++
#include <ros/ros.h>
#include <pcl_ros/point_cloud.h>
#include <pcl/point_types.h>
#include <pcl/point_types.h>
#include <pcl_conversions/pcl_conversions.h>
#include <sensor_msgs/PointCloud2.h>

typedef pcl::PointCloud<pcl::PointXYZ> PointCloud;

int main(int argc, char** argv)
{
  ros::init (argc, argv, "pub_pcl");
  ros::NodeHandle nh;
  ros::Publisher pub = nh.advertise<PointCloud> ("points2", 1);

  PointCloud::Ptr msg (new PointCloud);
  msg->header.frame_id = "some_tf_frame";
  msg->height = msg->width = 1;
  msg->points.push_back (pcl::PointXYZ(1.0, 2.0, 3.0));

  ros::Rate loop_rate(4);
  while (nh.ok())
  {
    pcl_conversions::toPCL(ros::Time::now(), msg->header.stamp);
    pub.publish (msg);
    ros::spinOnce ();
    loop_rate.sleep ();
  }
}
```

wkentaro, 2018-04-24:
I updated the wiki: http://wiki.ros.org/action/diff/pcl_ros?action=diff

closed 2018-04-24 by wkentaro