ros-perception/perception_pcl #189: Release for ROS melodic

hrnr, 2018-04-26:
This is quite a core stack, it would be nice to have this released early, so we could test applications on melodic.

wkentaro, 2018-04-26:
I commented to request the maintainer for release: https://github.com/ros-perception/pcl_conversions/issues/28#issuecomment-384687756.
And I added the test on Travis #190 to check if current default branch (lunar-devel) can be released.

paulbovbel, 2018-04-30:
Moved pcl_conversions to this repo (https://github.com/ros-perception/pcl_conversions/issues/28).

We need to make sure to sync up version numbers across packages to 1.6.0 on release into melodic.

wkentaro, 2018-04-30:
Can I release to the same code to lunar, kinetic, and indigo for sync?

paulbovbel, 2018-04-30:
We shouldn't release the same code-line for the other distros. At the very least, moving the location of the pcl_conversions package will make things messy.

We can backport features as necessary/desired to older *-devel branches, and cut releases from there.

wkentaro, 2018-04-30:
Ok, I see. Thanks!

wkentaro, 2018-04-30:
Can I close this?

paulbovbel, 2018-04-30:
Yes, thanks!

closed 2018-04-30 by paulbovbel