ros-perception/perception_pcl #205: color incorrect by convert_pointcloud_to_image

isbdnt, 2018-05-13:
the color of the model is orange, but the converted color is blue.
[image]
[image]

isbdnt, 2018-05-13:
this might be a issue of libgazebo_ros_openni_kinect.so. I printed out the PointXYZRGB from the topic published by libgazebo_ros_openni_kinect.so. the rgb field seems wrong.

wkentaro, 2018-05-16:
> this might be an issue of libgazebo_ros_openni_kinect.so. I printed out the PointXYZRGB from the topic published by libgazebo_ros_openni_kinect.so. the rgb values seemed wrong.

Yeah, I think so. Could you open an issue in https://github.com/ros-simulation/gazebo_ros_pkgs and close this issue?

SteveMacenski, 2019-11-13:
Closing, not a PCL Perception issue.

closed 2019-11-13 by SteveMacenski