ros-perception/perception_pcl #210: Download of table_scene_lms400.pcd during build

OkkeHendriks, 2018-06-13:
I am building ROS, including pcl_ros, in a production environment. 
In order to guarantee stable reproducible builds we cannot rely on build-time downloads.

Therefore I suggest to remove the following download from the build, at least make it optional:

https://github.com/ros-perception/perception_pcl/blob/12482e555dfc295c47dba1e7c14560870c6a6d0e/pcl_ros/CMakeLists.txt#L201

Also I notice that it is downloading directly into the source dir, shouldn't this be the build dir (CMAKE_CURRENT_BINARY_DIR)?
This way a .gitignore entry is needed for this file?

paulbovbel, 2018-06-14:
Good catch.

Looks like we added a sample launch file (i.e. a test) in this PR (https://github.com/ros-perception/perception_pcl/pull/147). Although the test is only run within the `if(CATKIN_ENABLE_TESTING)` block, the download is registered in the parent scope. It would make sense to move the download to the testing block as well.

OkkeHendriks, 2018-06-14:
Yes, inside the test block would already be better. 
We could also add the file to the repository -> no (extra) download required.
Or is this file considered binary/too large (17MB)?

paulbovbel, 2018-06-14:
I think Git LFS is the preferred approach for something like that, but I don't think it's supported on the upstream buildfarm. @nuclearsandwich could you please confirm the correct approach here?

nuclearsandwich, 2018-06-14:
> We could also add the file to the repository -> no (extra) download required.
> Or is this file considered binary/too large (17MB)?

17MB isn't enormous. Big binary files with plain git only start to cause real problems when they change often. This file doesn't appear to've changed since 2010 so I think adding it to the repository directly is worth considering. It's probably what I'd do here.

If there is a high concern about adding 17M to every new clone of the repository then the suggestion above to move the download block to CATKIN_ENABLE_TESTING will work. Is there a problem with builds failing when the download fails? Maybe there's a way to make download failure survivable and gracefully abort the test if the file isn't found?

> I think Git LFS is the preferred approach for something like that, but I don't think it's supported on the upstream buildfarm.

Regarding Git-LFS on the buildfarm, it's definitely not supported for debian/binary packaging jobs and I think it'd be possible but very messy to use it for CI/devel jobs. It's also important to bear in mind that Git-LFS requires an out-of-band file server as well and so it's not much more reproducible in theory than running a checksum on the file downloaded from its current location.

paulbovbel, 2018-06-14:
:+1: thanks, commit direct to repo sounds fine then, I'd welcome a PR @OkkeHendriks

OkkeHendriks, 2018-08-22:
No that is not an issue @paulbovbel, actually what I did locally, because I do not use the tests.
 I still agree with nuclearsandwich that adding it to the repo (and putting it in CATKIN_ENABLE_TESTING) is the best solution in this case.

paulbovbel, 2018-08-23:
I this we may not be on one page, because putting the download behind CATKIN_ENABLE_TESTING and adding the file to the repo, are mutually exclusive options.

Not a huge fan of making the base repository checkout 17 MB larger, forever. Since we don't have access to git-lfs, `catkin_download` seems like a sane alternative that's used extensively in other ROS projects. Just make sure to run your builds with `-DCATKIN_SKIP_TESTING=1`, and all concerns will be satisfied.

OkkeHendriks, 2018-10-11:
You are right :)

SteveMacenski, 2019-10-15:
Is sounds like this came to a good resolution :+1:  closing ticket!

closed 2019-10-15 by SteveMacenski