ros-perception/perception_pcl #216: keep_organized parameter makes no difference in pcl/CropBox nodelet

vbschettino, 2018-10-22:
Hi,

I think the title is self-explanatory; when using the pcl/CropBox nodelet, the keep_organized parameter seems to make no difference and the output cloud is always unorganized. Following is the launch file I used to see this:
```
<launch>
    <node pkg="nodelet" type="nodelet" name="pcl_manager" args="manager" output="screen" />

    <node pkg="nodelet" type="nodelet" name="crop_box" args="load pcl/CropBox pcl_manager" output="screen">
        <remap from="~input" to="/camera/depth_registered/points"/>
        <rosparam>
              keep_organized: true
        </rosparam>
    </node>
</launch>
```

SteveMacenski, 2019-10-18:
@vbschettino are you sure your pointcloud was organized to begin with? 

https://github.com/ros-perception/perception_pcl/blob/12482e555dfc295c47dba1e7c14560870c6a6d0e/pcl_ros/src/pcl_ros/filters/crop_box.cpp#L84

This logic looks pretty solid to me. What's your distribution / version?

mvieth, 2019-11-16:
I agree with @SteveMacenski that the `keep_organized` parameter is definitely passed on to the filter. I tested it (with an organized cloud) and found the same problem as @vbschettino. To me it seems that there is a bug in pcl itself, so I created an issue there.

SteveMacenski, 2019-11-16:
Awesome! Thanks for hunting that down! 

Closing this ticket and I’ll look at your new one

closed 2019-11-16 by SteveMacenski

peci1, 2022-01-09:
Just to explain the current status: Melodic is affected and [won't be fixed](https://github.com/PointCloudLibrary/pcl/issues/3471#issuecomment-602025794). Noetic is already fixed.

If anyone needs a working cropbox filter in Melodic, you can take inspiration here: https://github.com/tpet/nav_utils/blob/99b017599ba48e708d80b2391370f2d74169c227/src/nav_utils/crop_box_nodelet.cpp .