ros-perception/perception_pcl #219: 'struct pcl::PointCloud2' has no member named 'points'

GrantDare, 2019-01-17:
I was [recommended](https://answers.ros.org/question/312336/how-to-filter-pointcloud2-data-based-on-distances/?answer=312364#post-id-312364) to use pcl_ros to perform very basic filtering on a problem I am experiencing. The recommendation did not show any includes and for the life of my I am unable to solve the compile issue shown in the title, _'struct pcl::PointCloud2' has no member named 'points'_. 

If you could please assist in directing me to the correct includes required for the recommendation given, I would be very grateful.

Thank you in advance

thecoldviews, 2019-09-26:
try 'data' instead of 'points'

SteveMacenski, 2019-10-15:
Hi,
You're more likely to get help on library use from ROS Answers or reading the autogenerated docs. I'm going to close this ticket.

closed 2019-10-15 by SteveMacenski