ros-perception/perception_pcl #220: No rule to make target

AdamTheCanadian, 2019-01-31:
I am trying to build this package from source, cloned it into my workspace. When I try to build i get:

make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libpcl_common.so', needed by '/home/adam/catkin_ws/devel/lib/pcl_ros/convert_pointcloud_to_image'.  Stop.

However the file exists:
stat /usr/lib/x86_64-linux-gnu/libpcl_common.so
  File: /usr/lib/x86_64-linux-gnu/libpcl_common.so -> libpcl_common.so.1.8
  Size: 20        	Blocks: 0          IO Block: 4096   symbolic link
Device: 804h/2052d	Inode: 2770690     Links: 1
Access: (0777/lrwxrwxrwx)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2019-01-31 10:05:41.408792284 -0700
Modify: 2018-09-10 00:22:16.000000000 -0600
Change: 2019-01-30 09:48:45.358051267 -0700

Any suggestions?

paulbovbel, 2019-01-31:
None, sorry. Can't really give you any advice without further information about your environment, workspace, or what command you're running.

AdamTheCanadian, 2019-01-31:
> None, sorry. Can't really give you any advice without further information about your environment, workspace, or what command you're running.

I added some more information on my environment. Not sure what other information you require, let me know and I can provide it.

SteveMacenski, 2019-10-15:
@AdamTheCanadian is this still a concern?

SteveMacenski, 2019-11-13:
Closing, if still an issue feel free to comment and I'll reopen if still an issue.

closed 2019-11-13 by SteveMacenski