ros-perception/perception_pcl #222: Using pcl::transformPointCloud with melodic and tf2

Patrick-Lascombe, 2019-02-13:


peci1, 2019-03-06:
You can work around this using Eigen transform matrix. However, I also don't feel like this is the Melodic way of doing things...

    #include <tf2_eigen/tf2_eigen.h>
    const auto scan_tf = tf_buffer.lookupTransform(robot_frame,
         scan.header.frame_id, scan.header.stamp, ros::Duration(1));
     point_cloud.header.frame_id = robot_frame;
     pcl_ros::transformPointCloud(
         tf2::transformToEigen(scan_tf.transform).matrix(),
         scan, point_cloud);

peci1, 2019-03-06:
Depends on #171 ,

paulbovbel, 2019-03-06:
Sorry I didn't respond earlier. You're right, this is a bit of a mess. `tf2` operates on ROS message types, and has native support for `sensor_msgs::PointCloud2` via https://github.com/ros/geometry2/blob/melodic-devel/tf2_sensor_msgs/include/tf2_sensor_msgs/tf2_sensor_msgs.h

peci1, 2019-03-06:
This certainly is an option, but there still are the classes `tf2::Transform` etc. which might be present in this API, couldn't they?

paulbovbel, 2019-03-06:
absolutely, or more usefully the `tf2::doTransform` API should be defined for the pcl::PointCloud types.

peci1, 2019-04-26:
Solution proposed in https://github.com/ros-perception/perception_pcl/pull/231 .

closed 2019-10-15 by SteveMacenski

atyshka, 2019-10-15:
This was merged in April but I still don't see the new functions in the latest melodic binaries...

SteveMacenski, 2019-10-15:
That makes sense https://github.com/ros/rosdistro/blob/master/melodic/distribution.yaml#L5575 that release didn't make it to bloom.

I can take care of it

paulbovbel, 2019-10-16:
Sure thing. Just for sanity, @mjcarroll, could we please add Steve to PCL Team in ros-gbp?

mjcarroll, 2019-10-16:
I believe that I lack the power there, I'll ping someone who can.

nuclearsandwich, 2019-10-16:
Invite dispatched.

SteveMacenski, 2019-10-16:
Oh hi, got that email out of the blue and didn't know why I was added. 

Cool :+1:

atyshka, 2019-10-18:
Thanks for working on this! ROS Index said the version was updated to 1.7 on Oct 16 but Ubuntu 18.04 apt still has 1.6.2 as latest version. Is the ubuntu PPA only updated periodically?

SteveMacenski, 2019-10-18:
The last sync was about 3 weeks ago, it'll be a few more weeks before the next Melodic sync

peci1, 2019-10-18:
@atyshka Watch https://github.com/ros/rosdistro/pull/22689 . Once it's merged, wait for CI job http://build.ros.org/job/Mbin_uB64__perception_pcl__ubuntu_bionic_amd64__binary/ to re-run. When it finishes, the built .deb files will be uploaded into the shadow-fixed repo: http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-melodic-perception-pcl/ . And a few weeks later they will become available for everyone (when the so called "Melodic sync" happens, which will be announced on discourse.ros.org).

If you don't want to wait for the melodic sync and also don't want to add the whole shadow-fixed repo to your system, you can download the .deb files from shadow-fixed web and install them manually via `sudo dpkg -i package.deb`.