ros-perception/perception_pcl #228: Bug in transform.hpp?

PetrichSchmoin, 2019-04-05:
Hello everyone,

is there a bug in setting the `frame_id` of the output cloud in transform.hpp? I always get an empty `frame_id` for the output cloud when I call `transformPointcloud(..)`

In the transform.hpp at the lines 183 to 186 (and 216 - 219)
```
  cloud_out.header.frame_id = target_frame;  // <-- here the frame is set
  std_msgs::Header header;
  header.stamp = target_time;
  cloud_out.header = toPCL(header); // <-- and will be overitten here by an empty frame_id
```

Can anybody confirm this?

My setup:
Ubuntu 16.04
ROS Kinetic 
PCL 1.7

LucidOne, 2019-04-06:
I updated your post with links to the code, as I understand it.
This looks like a bug, and it should also be an easy unit test to write.
PRs as always are appreciated.

LucidOne, 2019-04-06:
The test should be something like [this](https://github.com/ros-perception/perception_pcl/blob/melodic-devel/pcl_conversions/test/test_pcl_conversions.cpp)

PetrichSchmoin, 2019-04-06:
I don't have much time to spend at the moment. Maybe I can try to fix it next month if no one else fixed it until then.

SteveMacenski, 2019-11-13:
@PetrichSchmoin any update? I would merge that PR pretty quickly

mvieth, 2019-11-15:
I created a PR. There is no unit test, but the problem itself should be fixed.

closed 2019-11-26 by SteveMacenski