ros-perception/perception_pcl #230: catkin_make error

npd26, 2019-04-25:
hello,

I am trying to use 

> perception_pcl

 package for realsense camera.

ROS: kinetic
Ubuntu: 16.04

after cloning package when I try to run catkin_make it is giving me error like,

`this file require compiler and library support for the ISO c++ 2011 standard.`

`std::this_thread has not been declared`

`std::chrono has not been declared`

how to solve this?

thanks.

LucidOne, 2019-04-26:
Please run `git branch` and check that you are using `kinetic-devel` or upgrade [gcc](https://gcc.gnu.org/wiki/C11Status) if you are attempting to backport `melodic-devel`.

npd26, 2019-04-28:
@LucidOne I am using git, kinetic devel but still not working. same error.

LucidOne, 2019-04-29:
Can you check that your copy [matches](https://github.com/kejxu/perception_pcl/blob/kinetic-devel/pcl_ros/tools/pcd_to_pointcloud.cpp)? What file references `std::chrono`?

npd26, 2019-05-08:
so, directly downloading zip file for kinetic-devel worked.

closed 2019-05-08 by npd26