ros-perception/perception_pcl #242: melodic-devel branch fails to build on Ubuntu 18.04/Melodic + PCL 1.9 with to ConvexHull error

patbenatar, 2019-07-31:
Trying to use `catkin build` to compile the `melodic-devel` branch on a Jetson Xavier with Ubuntu 18.04/Melodic + PCL 1.9 installed. I've traced this down to what I think is an issue related to qhull and an `#ifdef HAVE_QHULL` in `convex_hull.h` inside PCL, however I've confirmed that `libqhull-dev` is installed on the device...

```
In file included from /home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/src/pcl_ros/surface/convex_hull.cpp:40:0:
/home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/include/pcl_ros/surface/convex_hull.h:79:12: error: ‘ConvexHull’ in namespace ‘pcl’ does not name a template type
       pcl::ConvexHull<pcl::PointXYZ> impl_;
            ^~~~~~~~~~
/home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/src/pcl_ros/surface/convex_hull.cpp: In member function ‘void pcl_ros::ConvexHull2D::input_indices_callback(const PointCloudConstPtr&, const PointIndicesConstPtr&)’:
/home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/src/pcl_ros/surface/convex_hull.cpp:154:3: error: ‘impl_’ was not declared in this scope
   impl_.setInputCloud (cloud);
   ^~~~~
/home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/src/pcl_ros/surface/moving_least_squares.cpp: In member function ‘void pcl_ros::MovingLeastSquares::config_callback(pcl_ros::MLSConfig&, uint32_t)’:
/home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/src/pcl_ros/surface/moving_least_squares.cpp:216:48: warning: ‘void pcl::MovingLeastSquares<PointInT, PointOutT>::setPolynomialFit(bool) [with PointInT = pcl::PointXYZ; PointOutT = pcl::PointNormal]’ is deprecated: [pcl::surface::MovingLeastSquares::setPolynomialFit] setPolynomialFit is deprecated. Please use setPolynomialOrder instead. [-Wdeprecated-declarations]
     impl_.setPolynomialFit (use_polynomial_fit_);
                                                ^
In file included from /home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/include/pcl_ros/surface/moving_least_squares.h:44:0,
                 from /home/nvidia/catkin_ws/src/perception_pcl/pcl_ros/src/pcl_ros/surface/moving_least_squares.cpp:39:
/usr/local/include/pcl-1.9/pcl/surface/mls.h:362:7: note: declared here
       setPolynomialFit (bool polynomial_fit)
       ^~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/pcl_ros_surface.dir/src/pcl_ros/surface/convex_hull.cpp.o] Error 1
```

mvieth, 2019-11-06:
Is this still an issue?
If yes, you could try [this tutorial](http://www.pointclouds.org/documentation/tutorials/hull_2d.php) (switch out the `ConcaveHull` for `ConvexHull`). If the same error appears, the problem is with PCL itself and you should create an issue there.
It would also be interesting to know how you installed PCL. Since the apt-version for 18.04 is 1.8.1 and not 1.9, you probably compiled it yourself? Maybe some misconfiguration happened there?

patbenatar, 2019-11-11:
@mvieth I can't confirm if this is still an issue, as I left it off with a workaround and haven't gotten around to trying other solutions since then. We were compiling PCL from source, so given that I'd be OK with closing out this issue and considering it an edge case or an issue unrelated to this repository. Cheers :beers:

closed 2019-11-11 by patbenatar