ros-perception/perception_pcl #245: Missing Attributes in PCD file generated from bag to pcd

Dinatorehanova, 2019-08-25:
Hi,
I am using pcl_ros library's bag to pcd command to convert my pointcloud2 topic into pcd file. 

[image]

The PCD files should have the values for (range, reflectivity, timestamp, noise, x,y,z and itensity). However, when I am opening the PCD file in the cloudcompare I am getting a constant value of 0 for my range, reflectivity, noise, ring, and timestamp. And also many of my point's x,y,z are coming up with same value (0,0,0)

[image]

This is our bag file
https://drive.google.com/file/d/1a07ZoVyITabAu39hr9EVeHMpWqmCRVSd/view?usp=sharing

Here is a sample of pcd file generated from the "rosrun pcl_ros bag_to_pcd <input_file.bag> <topic> <output_directory>"
https://drive.google.com/file/d/1ZI9hUIjoYdnPh_crXvKg-fJcRqrTSzXJ/view?usp=sharing

Here is the head of my pcd file

[image]

Any help will be very great. We have been trying to resolve this issue for a long time.We are collecting data using an ouster LiDAR in ros bag.  

Thanks

SteveMacenski, 2019-10-16:
PRs welcome, my guess is the script doesn't support the full gambit of pointcloud template options but it shouldn't be too hard to add them

mvieth, 2019-11-12:
@Dinatorehanova The google drive links don't work anymore, would you provide new ones?
The issue is likely in `pcl_conversions.h`, maybe the field descriptions are not copied correctly, but that is difficult to say without the files.