ros-perception/perception_pcl #250: pcl_transformpointcloud cost high CPU

yp233, 2019-11-18:
I transform the pointcloud in the lidar frame to car frame in the callback by pcl_ros::transformpointcloud . It turns out that it cost too much CPU. I set the node frequency as 20 and the sensor frequency is 10. Am I wrong？

SteveMacenski, 2019-11-18:
Please post your code, what you expect, and specifically what happens. “Too much cpu” isn’t enough information. Amounts, what your baseline is, etc

yp233, 2019-11-19:
Sor, the code is private.So forgive me that i can't post the code . In the code i started a new thread for transforming the pointcloud in lidar frame to pointcloud in car frame. 
And the code is " pcl::transformPointCloud (scan_msgs_cloud_, cloud_tmp, sensor_in_base_);" 
The type of sensor_in_base_[pair.first] is eigen::martix4d. The amount of pointcloud is nearly 7000. And this code costs 10% CPU.

SteveMacenski, 2019-11-19:
Unfortunately without code samples or concrete details of an actual bug (used to be X% now X+10% after ABC), we can’t help.

closed 2019-11-19 by SteveMacenski

SteveMacenski, 2019-11-19:
The best advise I can give is to profile the bit of code and see where the time is being taken and then find if that’s actually an unreasonable amount of time for it to take for your unspecified platform

yp233, 2019-11-19:
I post code to your email. please check.

yp233, 2019-11-19:
@SteveMacenski 
Have you got the email ? 
Thanks a lot !

SteveMacenski, 2019-11-19:
@yp233 if you're not willing to post this code publicly, please do not send it to me. I'm not interested in getting in any IP issues with whatever company or group you're working with. 

Please do the appropriate about of due diligence to fully describe your issue and determine if its actually a bug or that you're trying to do too much on a limited compute platform. I could believe with all the PCL changes that things need some tweaking, but I'm not going to have time to try to guess what your issue is.

Do some benchmarking, if you had an example of code or system that has less CPU that makes you believe this is too much, post diffs or commit hashes. Profile the code to see where the time is going. Try with other pointclouds or sensors to see how it changes.

yp233, 2019-11-19:
Thanks a lot ！ I will do more test by myself. Thanks for ur attention.