ros-perception/perception_pcl #259: dont support pcl-1.9 ???

chennuo0125, 2020-01-02:
/home/ubuntu/catkin_ws/src/perception_pcl/pcl_ros/include/pcl_ros/point_cloud.h:79:37: error: no match for ‘operator=’ (operand types are ‘std::shared_ptr<std::vector<pcl::detail::FieldMapping> >’ and ‘boost::shared_ptr<std::vector<pcl::detail::FieldMapping> >’)
       pcl::detail::getMapping(*msg) = mapping_;

SteveMacenski, 2020-01-02:
Can you please add some detail on what's going on.

Posting a question and what apperas to be an error next to a line of code doesn't really give much context or ability to assist .

chennuo0125, 2020-01-02:
my system is ubuntu kinetic. i want to compile a pkg depended on pcl_ros pkg, but i encounter error as follows:
[image]
the error maybe is caused by that i have installed pcl-1.9.1, so i want to install pcl_ros by source code, but i encounter another problem as follows:
[image]

SteveMacenski, 2020-01-02:
The officially supported PCL version for Kinetic is 1.7.x (https://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021) 

[This](https://github.com/ros-perception/perception_pcl/pull/258)) PR would probably fix your issue. 1.9 has API changes that are incompatible with this package in Kinetic.

taketwo, 2020-01-09:
It appears that you have installed PCL from the `master` branch, i.e. not `1.9.1`, but rather `1.9.1-dev`. It is indeed incompatible with this package. Your options are either apply this patch or install released PCL 1.9.1.

SteveMacenski, 2020-05-05:
Closing from solution provided by @taketwo

closed 2020-05-05 by SteveMacenski