ros-perception/perception_pcl #278: Foxy release

jacobperron, 2020-05-12:
As we prepare for the release of Foxy, the `perception_pcl` packages should be released as they are [part of `desktop` variant](https://github.com/ros2/variants/blob/c0f3fce1afed398a432f4262e0887e57ff37c61c/desktop/package.xml#L17).

SteveMacenski, 2020-05-12:
Noted - I'll put on my queue for tomorrow now that the 1.11 changes are in.

SteveMacenski, 2020-05-12:
@jacobperron we still need pcl msgs. I'm submitting that now but that needs to be in before I can get CI spinning on Foxy for this repo to release it.

SteveMacenski, 2020-05-13:
Release https://github.com/ros-perception/perception_pcl/releases/tag/2.2.0

https://github.com/ros/rosdistro/pull/24871

This covers PCL conversions, PCL ROS is still unported. 

PS, I know there's some blocking chart for release from Noetic I can't seem to find one for Foxy. Do you know where the tool is to see what's blocking what for release?

closed 2020-05-13 by SteveMacenski

jacobperron, 2020-05-13:
Thanks!

> PS, I know there's some blocking chart for release from Noetic I can't seem to find one for Foxy. Do you know where the tool is to see what's blocking what for release?

I think you are referring to this: http://repo.ros2.org/status_page/blocked_releases_foxy.html?

SteveMacenski, 2020-05-13:
:facepalm: I tried changing just the noetic to foxy on this URL I have saved http://repositories.ros.org/status_page/blocked_releases_noetic.html?s=6&r=1

nnmm, 2020-10-08:
@SteveMacenski Our group is porting a ROS1 repo which uses `pcl_ros` to ROS2. Ideally we'd like to use a precompiled package for Foxy. I assume your comment in https://answers.ros.org/question/359821/how-to-use-pcl_ros-with-ros-2-eloquent/ still applies and the migration is not complete yet? If you could summarize the status and what still needs to be done, that would be very helpful for us to see if we can already use it in some form and whether it makes sense for us to help with porting.

cc @fred-apex-ai

SteveMacenski, 2020-10-08:
The 2 draft, incomplete PRs from older ROS2 distributions is all we have right now. So far, no one's wanted to put in the time to complete those ports or port things. I don't think we need to port things all at once, so just sections of it at a time.