ros-perception/perception_pcl #283: PCLNodelet: The tf listener doesn't use the nodelet nodehandles.

peci1, 2020-06-11:
The `tf_listener_` member of `PCLNodelet` is only default-constructed, which means it uses a default-constructed NodeHandle. That means no `nodelet load` remappings have an effect on the topics the TF listener subscribes to. Only remappings done on the manager will affect it. I think the listener should use the nodelet's dedicated nodehandles, which correctly receive the needed remappings.

MWE:

```
$ rosrun nodelet nodelet manager __name:=m &
$ rosrun nodelet nodelet load pcl/CropBox m /tf:=/asd pcl_CropBox/output:=aaa &
$ rosnode info -q m
--------------------------------------------------------------------------------
Node [/m]
Publications: 
 * /aaa [sensor_msgs/PointCloud2]
 * /m/bond [bond/Status]
 * /pcl_CropBox/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /pcl_CropBox/parameter_updates [dynamic_reconfigure/Config]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /m/bond [bond/Status]
 * /tf [unknown type]
 * /tf_static [unknown type]
...
```

As you can see, the output topic got remapped correctly, while the tf topic remained untouched.

The best solution from my point of view would be to convert the listener into a unique_ptr and initialize it in the `onInit()` method. But that would be a big ABI and API break.

I found a workaround using placement-new in the beginning of `PCLNodelet::onIinit()`, which looks a little bit hacky, but does its job:

```
const auto cacheLength = tf_listener_.getCacheLength();
tf_listener_.~TransformListener();
new (&tf_listener_) tf::TransformListener(getNodeHandle(), cacheLength);
```

SteveMacenski, 2020-06-11:
Its unideal to rely on something like that. The only reason you have the ability to remap that easily is due to the use of TF. TF2 removed the ability to do that. TF is deprecated in Melodic but still released but in Noetic and ROS2 that ability doesn't exist. I don't think you should rely on a deprecated feature.

peci1, 2020-06-12:
I'm sorry, but I don't think you're getting me right. I'm not "remapping" TF frames. I'm remapping topics. What would be deprecated about remapping (general) topics?

SteveMacenski, 2020-06-12:
Yeah, that's actually the issue. If I understand you correctly [these docs](http://wiki.ros.org/tf2) show deprecation in TF2. e.g.

```
 No tf_prefix: In previous versions there was a concept of a tf_prefix which would be prepended to the frame name using a / separator. A leading slash used to indicate that it had already been prefixed. For backwards compatibility tf2 will strip any leading / character.

No Remapping:

The concept of tf frame_ids is not scoped in the same way as ROS Names. In particular, namespacing a specific subpart of a computation graph does not change the physical layout which the tf tree represents. Because of this frame_ids do not follow namespace remapping rules. It is common to support a ROS parameter to allow changing frame_ids used in algorithms. 
```

More info: https://www.osrfoundation.org/wordpress2/wp-content/uploads/2015/04/tf2.pdf

They hardcode `/tf` into the code and make it very difficult to remap that topic in TF2. So I think what you're trying to do is woefully deprecated so in fixing that issue, its not really a "fix" since TF1 doesn't exist in ROS2 or Noetic (or even officially in Melodic, I don't believe)

peci1, 2020-06-15:
We're still talking about two different remappings.

What tf2 stopped supporting is remapping TF frame_ids, so e.g. frame `base_link` will not get changed if you launch the node with remapping args `base_link:=other_link`.

What I'm talking about is remapping the name of the `/tf` topic itself, which is okay and supported throughout ROS (it is just a normal topic, nothing special about that). It just tells the TF listeners and broadcasters to take transforms from a topic with a different name. This comes pretty handy in many cases - e.g. if you want to run a TF server concentrating data from all TF publishers and offering them as a single publisher. This architecture effectively decreases the M:N connection between M publishers and N subscribers to M:1 and 1:N. And there are other uses, e.g. sending a subsampled TF topic over slow network, and so on.

SteveMacenski, 2020-06-15:
> What I'm talking about is remapping the name of the /tf topic itself

That's also what I'm talking about. That is not remappable in TF2. Part of no tf prefixes is also that there's not recommended behavior to rename the TF topic or add namespaces. Please review the TF2 and migration guide documentation.

paulbovbel, 2020-06-15:
It's totally normal to remap `/tf*` topics, such as for segregating multiple tf trees for multi-robot setups. Transport topics are completely distinct from frame ids.

paulbovbel, 2020-06-15:
AFAIK it's totally normal to remap `/tf*` topics, e.g. for partitioning tf trees for multi-robot setups. Transport topics are completely distinct from frame ids, and nothing prevents, or advises against, remapping the former.

paulbovbel, 2020-06-15:
AFAIK it's totally normal to remap `/tf*` topics, e.g. for partitioning tf trees as @peci1 calls out. Transport topics are completely distinct from frame ids, and nothing prevents, or advises against, remapping the former.