ros-perception/perception_pcl #286: [foxy] pcl_conversions error: PCL requires C++14 or above

ShivamPR21, 2020-07-01:
I have manually compiled pcl version 1.11 and while building pcl_conversions with it, throws an error repeatedly saying PCL requires C++14 or above as discussed in https://github.com/PointCloudLibrary/pcl/issues/3104 it seems that  target should link against PCL_LIBRARIES like 
```target_link_libraries(${PROJECT_NAME}-test ${Boost_LIBRARIES} ${PCL_LIBRARIES})``` 
in pcl_conversions/cmakelist.txt this worked for me is there something else reason behind the error.

The full output is attached here :- 
[stdout_stderr.log](https://github.com/ros-perception/perception_pcl/files/4856963/stdout_stderr.log)

Here is my stderr log snippet:-
```-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.5 (/home/shivam/ros2_foxy/install/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- Found message_filters: 3.2.4 (/home/shivam/ros2_foxy/install/message_filters/share/message_filters/cmake)
-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.7
-- Found Eigen: /usr/include/eigen3 (Required is at least version "3.1") 
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Found USB_10: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- OpenNI2 found (include: /usr/include/openni2, lib: /usr/lib/libOpenNI2.so)
-- Found libusb-1.0: /usr/include  
-- Found PCL_COMMON: /home/shivam/common_pkgs/install/pcl/lib/libpcl_common.so  
-- Found PCL_OCTREE: /home/shivam/common_pkgs/install/pcl/lib/libpcl_octree.so  
-- Found PCL_IO: /home/shivam/common_pkgs/install/pcl/lib/libpcl_io.so  
-- Found pcl_msgs: 1.0.0 (/home/shivam/ros2_perception/install/pcl_msgs/share/pcl_msgs/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_introspection_c;rosidl_typesupport_fastrtps_c
-- Found rosidl_adapter: 1.0.1 (/home/shivam/ros2_foxy/install/rosidl_adapter/share/rosidl_adapter/cmake)
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.1.1f")  
-- Found FastRTPS: /home/shivam/ros2_foxy/install/fastrtps/include  
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp;rosidl_typesupport_fastrtps_cpp
-- Found rclcpp: 2.0.0 (/home/shivam/ros2_foxy/install/rclcpp/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 1.0.1 (/home/shivam/ros2_foxy/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found ament_cmake_gtest: 0.9.5 (/home/shivam/ros2_foxy/install/ament_cmake_gtest/share/ament_cmake_gtest/cmake)
-- Found gtest sources under '/home/shivam/ros2_foxy/install/gtest_vendor/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Configuring done
-- Generating done
-- Build files have been written to: /home/shivam/ros2_perception/build/pcl_conversions
[35m[1mScanning dependencies of target gtest_main[0m
[35m[1mScanning dependencies of target gtest[0m
[ 16%] [32mBuilding CXX object gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o[0m
[ 33%] [32mBuilding CXX object gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o[0m
[... 85 lines cut ...]
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:57:91:[m[K [01;31m[Kerror: [m[Kexpected primary-expression before ‘[01m[K>[m[K’ token
   57 |           std::make_unique<rclcpp::experimental::buffers::RingBufferImplementation<BufferT[01;31m[K>>[m[K(
      |                                                                                           [01;31m[K^~[m[K
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:62:16:[m[K [01;31m[Kerror: [m[K‘[01m[Kmake_unique[m[K’ is not a member of ‘[01m[Kstd[m[K’
   62 |           std::[01;31m[Kmake_unique[m[K<rclcpp::experimental::buffers::TypedIntraProcessBuffer<MessageT, Alloc,
      |                [01;31m[K^~~~~~~~~~~[m[K
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:62:16:[m[K [01;36m[Knote: [m[K‘[01m[Kstd::make_unique[m[K’ is only available from C++14 onwards
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:63:29:[m[K [01;31m[Kerror: [m[Kexpected primary-expression before ‘[01m[K>[m[K’ token
   63 |             Deleter, BufferT[01;31m[K>>[m[K(
      |                             [01;31m[K^~[m[K
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:74:16:[m[K [01;31m[Kerror: [m[K‘[01m[Kmake_unique[m[K’ is not a member of ‘[01m[Kstd[m[K’
   74 |           std::[01;31m[Kmake_unique[m[K<rclcpp::experimental::buffers::RingBufferImplementation<BufferT>>(
      |                [01;31m[K^~~~~~~~~~~[m[K
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:74:16:[m[K [01;36m[Knote: [m[K‘[01m[Kstd::make_unique[m[K’ is only available from C++14 onwards
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:74:91:[m[K [01;31m[Kerror: [m[Kexpected primary-expression before ‘[01m[K>[m[K’ token
   74 |           std::make_unique<rclcpp::experimental::buffers::RingBufferImplementation<BufferT[01;31m[K>>[m[K(
      |                                                                                           [01;31m[K^~[m[K
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:79:16:[m[K [01;31m[Kerror: [m[K‘[01m[Kmake_unique[m[K’ is not a member of ‘[01m[Kstd[m[K’
   79 |           std::[01;31m[Kmake_unique[m[K<rclcpp::experimental::buffers::TypedIntraProcessBuffer<MessageT, Alloc,
      |                [01;31m[K^~~~~~~~~~~[m[K
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:79:16:[m[K [01;36m[Knote: [m[K‘[01m[Kstd::make_unique[m[K’ is only available from C++14 onwards
[01m[K/home/shivam/ros2_foxy/install/rclcpp/include/rclcpp/experimental/create_intra_process_buffer.hpp:80:29:[m[K [01;31m[Kerror: [m[Kexpected primary-expression before ‘[01m[K>[m[K’ token
   80 |             Deleter, BufferT[01;31m[K>>[m[K(
      |                             [01;31m[K^~[m[K
```

ShivamPR21, 2020-07-01:
[image]

SteveMacenski, 2020-07-01:
https://www.ros.org/reps/rep-2000.html#foxy-fitzroy-may-2020-may-2023

The minimum language requirements for Foxy is C++14 so I think that's already fulfilled unless we're setting it somewhere in the code to a lesser value. If there's some problem with linking, a PR would be appreciated to resolve any issues. Remember that Foxy officially supports 1.10.0 not 1.11.0, but open to helping you get it working in that version as well.

ShivamPR21, 2020-07-01:
I think that this a linker error as it's written in pcl docs https://pcl.readthedocs.io/projects/tutorials/en/latest/using_pcl_pcl_config.html#using-pcl-pcl-config the target_link_library() is needed for telling compiler which libraries we are linking against this answer also tells the same https://stackoverflow.com/a/43457154.

ShivamPR21, 2020-07-01:
@SteveMacenski probably I may be wrong in deducing the actual cause of the error but does there any reason to deny target_link_libraries() for project.

SteveMacenski, 2020-07-01:
Please submit a PR

closed 2020-07-01 by SteveMacenski