ros-perception/perception_pcl #291: Potentially re-scope perception_pcl to perception_3d and include conversions for open3d

SteveMacenski, 2020-07-24:
Per https://github.com/unr-arl/open3d_ros/issues/6#issuecomment-663727821 we have some traction for including conversions from ROS to open3D as well as PCL. These are pretty simple functions, but having that as 'batteries included' in ROS would be great if more users decide to play around or use Open3D rather than PCL.

I'd like to propose renaming this repo `perception_3d` and maintain everything else. We would include `open3d_conversions` as a new package from @nkhedekar et. al from [here](https://github.com/unr-arl/open3d_ros).

CC @sloretz @clalancette any objections to that potential change? I believe the only downstream effects would be to update the URL for bloom when we do releases, but github is good about redirecting to renamed repositories. I don't think we should run into any issues otherwise since the package names will remain the same.

clalancette, 2020-07-25:
I will say that `perception_pcl` feels a lot more descriptive than `perception_3d`.  The latter could mean a lot of different things.

How about introducing a new repository `perception_open3d`?  That keeps the churn here minimal, keeps the names of the repositories generally descriptive, and keeps with the federated model of ROS.  Thoughts?

SteveMacenski, 2020-07-25:
I'd also be OK with that. Given the (very small) amount of code, having in the same repo is just convenient for releases. I'm not sure I follow the federated model comment - these are still independent packages to be installed separately. They'd just in the same repo for organizational purposes.

clalancette, 2020-07-25:
> I'd also be OK with that. Given the (very small) amount of code, having in the same repo is just convenient for releases. I'm not sure I follow the federated model comment - these are still independent packages to be installed separately. They'd just in the same repo for organizational purposes.

It just seems that separate repos for separate concerns makes sense to me.  On the other hand, if you consider that PCL and Open3D address the same concern, then renaming this repo to something like `perception_pcl_open3d` may make sense (I don't really like that name, but something that is a bit more descriptive would help).

Ultimately, I don't have a strong opinion.  As long as the package names aren't changing, it should be fairly painless from a ROS distribution point-of-view.

SteveMacenski, 2020-07-25:
Ok. I can either way on it too. Maybe just a new repo is the way to go if not strongly opinionated.

nkhedekar, 2020-07-27:
You're right, `perception_open3d` would be the better option. If someone can make the repository, @matnay and I can make a PR into it.

SteveMacenski, 2020-07-27:
https://github.com/ros-perception/perception_open3d done. I opened it under and Apache 2.0 license. If you'd like another license, let me know. Apache 2.0 is typically what we use for ROS-core-things unless you have a strong opinion on another one. Once you have the PR there, I'll work on adding CI and releasing it (would you like to know how to do that?). I'll add you as an admin as well to be able to OK/merge things.

nkhedekar, 2020-07-27:
Thank you @SteveMacenski! Yep, that'll work.

SteveMacenski, 2020-07-27:
Let me know in the CI all the github user names I should give admin for on your side

nkhedekar, 2020-07-27:
Yep, this would be @nkhedekar, @matnay and @kostas-alexis

closed 2020-07-27 by SteveMacenski