ros-perception/perception_pcl #294: [Eloquent] Could not find package "PCL"

ItsMeTheBee, 2020-08-03:
Hey there!

I´m a little confused about the installation of this package.
In my workspace (in a ros:eloquent docker container) I pulled this repo, checked out eloquent_devel and tried building it using 
`colcon build --symlink-install`

This results in 

```
Starting >>> pcl_conversions
--- stderr: pcl_conversions                         
CMake Error at CMakeLists.txt:17 (find_package):
  By not providing "FindPCL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PCL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PCL" with any of
  the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< pcl_conversions [0.84s, exited with code 1]

Summary: 0 packages finished [0.99s]
  1 package failed: pcl_conversions
  1 package had stderr output: pcl_conversions
  1 package not processed
```

I played around a bit and i think the issue is that i´m not able to build ros_pcl.

`colcon build --symlink-install --packages-select pcl_ros`

results in this message:

`WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'pcl_ros' in --packages-select`

Any idea how to fix this?

SteveMacenski, 2020-08-03:
Please ask on ROS Answers

closed 2020-08-03 by SteveMacenski

ItsMeTheBee, 2020-08-03:
I didn´t know PCL is a ubuntu package.
`apt install libpcl-dev` fixed the issue.