ros-perception/perception_pcl #295: Wiki page out-of-date

kunaltyagi, 2020-08-06:
Since PCL shifted to `std::shared_ptr`, the documentation on the wiki page is out of date. Example: https://wiki.ros.org/pcl_ros#Publishing_point_clouds

It uses a typedef from PCL, which is now `std::shared_ptr<pcl::PointCloud<pcl::PointXYZ>>` but ROS expects `boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ>>`

`ros_ptr` and `pcl_ptr` for interfacing between ROS and PCL might also need to be introduced.

kunaltyagi, 2020-08-06:
Relevant PointCloudLibrary/pcl#4307

SteveMacenski, 2020-08-06:
What specifically should change? I don't see anything with `boost::shared_ptr`, things look OK to me. Everything is typedefed from `typedef pcl::PointCloud<pcl::PointXYZ> PointCloud;`, which I think should hide any of that to a user?

kunaltyagi, 2020-08-06:
With PCL 1.11 at least, the following lines will throw an error:
* https://wiki.ros.org/pcl_ros#CA-226070dcbe48ef19c444eb4914084fc903868f16_14
* https://wiki.ros.org/pcl_ros#CA-2c92c01374035f603e08bcde0eb24382b1ccd02e_8

That's due to them using `std::shared_ptr` (from PCL) while ROS expected `boost::shared_ptr`

SteveMacenski, 2020-08-06:
What's the specific fix I should change it to

kunaltyagi, 2020-08-06:
* Not use `PointCloud<T>::{Const}Ptr` on subscriber callback parameter. Instead use `boost::shared_ptr<{const} PointCloud<T>>`
* Use `ros_ptr(message)` in the publish calls to convert from std pointer to boost pointerc

fawadahm, 2021-02-05:
@kunaltyagi: This has been very useful. I am trying to use ROS Melodic with PCL 1.11.0 via perception_pcl but I am facing the same problem you mentioned above when publishing a point cloud. I tried to use ```ros_ptr```. You can find the code as follows:

```
#include <ros/ros.h>
#include <sensor_msgs/PointCloud2.h>
#include <pcl_conversions/pcl_conversions.h>
#include <pcl_ros/point_cloud.h>
#include <pcl/point_types.h>
typedef pcl::PointCloud<pcl::PointXYZ> PointCloud;
void Publish_Cloud ( std::string frame_id, PointCloud::Ptr input_cloud )
{
  input_cloud->header.frame_id = frame_id;
  pub.publish ( pcl::ros_ptr ( input_cloud->makeShared () ) );
  // std::cout << "Published point cloud with frame_id: " << frame_id << std::endl;
}
```

However, I cannot find ```ros_ptr``` in the pcl namespace. Could you help me tackle this?
In ```catkin_ws/src/``` I have my ```perception``` package and the melodic_devel version of ```perception_pcl```

kunaltyagi, 2021-02-05:
It should be present at the end of [pcl_ros/point_cloud.h](https://www.github.com/ros-perception/perception_pcl/tree/melodic-devel/pcl_ros/include/pcl_ros/point_cloud.h)

fawadahm, 2021-02-05:
@kunaltyagi , thank you. I do see it there and I am trying to reference it using that namespace ```pcl::ros_ptr``` but I keep getting the same error i.e., ```ros_ptr is not a member of pcl```. Could you please help me dig deeper? The following is what my CMakeLists.txt looks like:

```
# %Tag(FULLTEXT)%
cmake_minimum_required(VERSION 2.8.3)
project(perception)

## Find catkin and any catkin packages
find_package(catkin REQUIRED COMPONENTS
	roscpp
	rospy
	std_msgs
	genmsg
	PCL
	rosbag)

## Generate added messages and services
generate_messages(DEPENDENCIES std_msgs)

## Declare a catkin package
catkin_package()

## Build talker and listener
include_directories(include 
					${catkin_INCLUDE_DIRS} 
					${PCL_INCLUDE_DIRS}
    				#${Boost_INCLUDE_DIRS}
					)

link_directories(${PCL_LIBRARY_DIRS})
add_definitions(${PCL_DEFINITIONS})

# We need boost libraries
find_library(BOOST_SERIALIZATION boost_serialization)
if (NOT BOOST_SERIALIZATION)
	message(FATAL_ERROR "Can't find libboost_serialization")
endif()

find_library(BOOST_SYSTEM boost_system)
if (NOT BOOST_SYSTEM)
	message(FATAL_ERROR "Can't find libboost_serialization")
endif()

# Lidar compression from a bag
add_executable(pc_publisher src/point_cloud_publisher.cpp)
target_link_libraries(pc_publisher ${catkin_LIBRARIES} ${BOOST_SYSTEM} ${PCL_LIBRARIES})# ${Boost_LIBRARIES})
add_dependencies(pc_publisher netdriving_perception)
```

kunaltyagi, 2021-02-05:
You need pcl_ros package, not the pcl catkin package. You might also need to use find package pcl and link the pcl libraries, if I remember correctly

fawadahm, 2021-02-05:
Thank you. By any chance, do you a catkin package that uses pcl_ros that I can use as a reference?

mvieth, 2021-02-05:
> Thank you. By any chance, do you a catkin package that uses pcl_ros that I can use as a reference?

Have a look at the [wiki page](https://wiki.ros.org/pcl_ros), on the right side, under "Used by".

fawadahm, 2021-02-05:
Got it, thanks for all the help. It seems like I had installed an older version of pcl_ros through apt-get that I needed to remove and build the melodic_devel branch of pcl_ros instead.