ros-perception/perception_pcl #307: [pcl_conversions / pcl_ros] Depending on libpcl-all-dev implicitly introduces UI dependencies

YoshuaNava, 2020-09-03:
Dear Developers,
My name is Yoshua Nava. I'm a Perception Software Engineer at ANYbotics AG (Zurich, Switzerland)

We use the `perception_pcl` packages in our software (specially `pcl_ros`), building upon the many cool tools from the Point Cloud Library. We install these in our robot, which has PCs that run in headless mode (no GUI).

However, we have lately noticed that when installing `pcl_ros` through the ROS PPA, it grabs X11 dependencies, Qt and graphics drivers. By backtracking the source tree we found that the depend `libpcl-all-dev`, which is a hook to the apt package `libcl-dev` [[link](https://packages.ubuntu.com/focal/libpcl-dev)], depends on vtk, which in turns depends directly on Qt and mesa. Thus, when installing software on the robot, we are asked to install graphic libraries in our headless PCs. Furthermore, these graphic libraries introduce extra tools, such as apport, which introduce conflicts that prevents us from using tools such as `systemd-coredump` for debugging.

We would like to have more fine-grained control over these dependencies, and keep depending on perception_pcl without having to install graphic dependencies. I sent a message to @SergioRAgostinho, one of the main developers of PCL, and he replied that he should rather have dependencies to individual packages here, for example:

```
libpcl-common
libpcl-conversions-dev
libpcl-features
libpcl-filters
libpcl-io
libpcl-kdtree
libpcl-keypoints
libpcl-ml
libpcl-msgs-dev
libpcl-octree
libpcl-outofcore
```

I would like to propose making this change to the corresponding packages of perception_pcl. Would the maintainers approve of this initiative? If affirmative I can open the corresponding pull requests to add the fine-grained pcl packages to [rosdistro](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml) and then to perception_pcl.

Thank you in advance.

Best regards,
Yoshua Nava

YoshuaNava, 2020-09-03:
@clalancette Thank you very much, that is great to hear.

I also got in contact with the rosdistro maintainers to introduce the individual packages to the rosdep keys. And I sent an email to the maintainers of the PCL PPA, because I found out that pcl-io also depends on graphical libraries: https://packages.ubuntu.com/focal/libpcl-io1.10

I have to wait for those two sources to reply before I proceed, but until that happens I can make a list of packages that we would need to depend to in perception_pcl.

**pcl_ros**

* libpcl-common
* libpcl-features
* libpcl-filters
* libpcl-io
* libpcl-kdtree
* libpcl-segmentation
* libpcl-surface

**pcl_conversions**

* libpcl-common
* libpcl-io

SteveMacenski, 2020-09-03:
I’ve got no issue with any of that, but close the PR until those rosdep keys are valid and could be reasonably merged.

YoshuaNava, 2020-09-03:
@SteveMacenski Oki.

I have submitted https://github.com/ros/rosdistro/pull/26448, will wait for it to be merged and then re-open the PR here.

SteveMacenski, 2020-09-03:
Thanks, I just don't want to have someone accidentally merge a very reasonable looking PR without knowing the keys don't exist :smile:

YoshuaNava, 2020-09-07:
Hi @SteveMacenski and @clalancette,

I'm currently finishing the PR at rosdistro, collecting individual PCL Debian package IDs for each Linux distro where ROS is deployed. I have also gotten in contact with the Ubuntu Debian maintainers to see if some of the UI depends can be split away from the modules such as `pcl-io`, which depends on vtk.

Until I'm finished, I was considering proposing that we replaced the `depend` of pcl_ros on libpcl-all-dev with a `build_depend` (suggested by @jspricke, maintainer of the PCL Ubuntu Debian repos). This would guarantee that the extra packages from PCL are only needed during build time, and not when deploying the software on a robot.

What do you think? Would you be ok with this proposal?

Thank you in advance.

Best regards,
Yoshua Nava

clalancette, 2020-09-07:
> Until I'm finished, I was considering proposing that we replace the `depend` of pcl_ros on libpcl-all-dev with a `build_depend` (suggested by @jspricke, maintainer of the PCL Ubuntu Debian repos). This would guarantee that the extra packages from PCL are only needed during build time, and not when deploying the software on a robot.

For `pcl_conversions`, there isn't anything to change; it is already a `build_export_depend` on libpcl-all-dev, which makes sense since that is a header-only library.

For `pcl_ros`, you could indeed split the current `<depend>libpcl-all-dev</depend>` into `<build_depend>libpcl-all-dev</build_depend>` and `<exec_depend>libpcl-all</exec_depend>`.

closed 2020-11-12 by YoshuaNava

noelex, 2022-11-02:
Sorry for resurrecting this but it looks like installing `pcl_ros` still cause installation of graphic packages.

Adding `<build_depend>libpcl-all-dev</build_depend>` and `<exec_depend>libpcl-all</exec_depend>` won't resolve this as `libpcl-all` declares dependency on `libpcl-visualization1.10` which depends on other graphic stuffs.

Currently I have to use `<build_depend>libpcl-all-dev</build_depend>` without `<exec_depend>libpcl-all</exec_depend>` and install required pcl runtime packages manually to workaround this.

clalancette, 2022-11-02:
> Sorry for resurrecting this but it looks like installing `pcl_ros` still cause installation of graphic packages.

Correct, the original reporter opened up https://github.com/ros/rosdistro/pull/26448 , but never followed through to finish it.  If you'd like to pick that up, it is one way to proceed.

mvieth, 2022-11-02:
I created https://github.com/ros/rosdistro/pull/35137 . Feel free to take a look.

In `pcl_ros`'s package.xml we could then use:
```
  <build_depend>libpcl-all-dev</build_depend>
  <build_export_depend>libpcl-all-dev</build_export_depend>
  <exec_depend>libpcl-common</exec_depend>
  <exec_depend>libpcl-features</exec_depend>
  <exec_depend>libpcl-filters</exec_depend>
  <exec_depend>libpcl-io</exec_depend>
  <exec_depend>libpcl-segmentation</exec_depend>
  <exec_depend>libpcl-surface</exec_depend>
```
instead of the current `<depend>libpcl-all-dev</depend>`