ros-perception/perception_pcl #309: How to use pcl_conversion to convert pointcloud2 massage (PiontXYZRGB) into PointXYZI

sonumathur, 2020-09-15:
I am using ROS to capture pointcloud using realsense D435 depth camera on Ubuntu 18.04 and ROS Melodic. The pointcloud produced using the "rosrun pcl_ros pointcloud_to_pcd input:=/" is PointXYZRGB type. I want to convert this pointcloud2  to PointXYZI using pcl_conversion.

How can i Convert this in ROS environment  ......?

mvieth, 2020-09-27:
The published clouds probably have the RGB field, and they are written without any conversion to the pcd file. You could either adapt the [pointcloud_to_pcd node](https://github.com/ros-perception/perception_pcl/blob/melodic-devel/pcl_ros/tools/pointcloud_to_pcd.cpp) to your needs, or deal with the conversion when you load the pcd files.

rafaelRinconFo, 2020-10-30:
Hi @sonumathur Did you find a way to do it?

lidiaxp, 2021-04-27:
[This](https://github.com/davidfornas/pcl_manipulation) package do it, I have teste on Ubuntu 18 and 20, it works.

lidiaxp, 2021-04-27:
gen = point_cloud2.read_points(data, skip_nans=True)
int_data = list(gen)

a, b, c = [], [], []

for x in int_data:
    a.append(round(x[0]))
    b.append(round(-x[1]))
    c.append(round(x[2]))

SteveMacenski, 2021-04-27:
Closing as answered (and also this isn't really a topic for this issue tracker, should be on ROS Answers or PCL forums)

closed 2021-04-27 by SteveMacenski