ros-perception/perception_pcl #312: unable to build pcl_ros

Magesh7803, 2020-10-21:
HI,
I have source installation of Foxy and while tryinf to build perception_pcl, i get the following error. Can you pls suggest what could be the issue.
ros2@ros2-HP-EliteBook-850-G3:~/tespcl$ colcon build
Starting >>> pcl_msgs
Finished <<< pcl_msgs [23.4s]                         
Starting >>> pcl_conversions
Finished <<< pcl_conversions [23.4s]                         
Starting >>> pcl_ros
--- stderr: pcl_ros                                    
<command-line>:0:1: error: macro names must be identifiers
make[2]: *** [CMakeFiles/pcl_ros_tf.dir/src/transforms.cpp.o] Error 1
make[1]: *** [CMakeFiles/pcl_ros_tf.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< pcl_ros [23.1s, exited with code 2]

Summary: 2 packages finished [1min 10s]
  1 package failed: pcl_ros
  1 package had stderr output: pcl_ros
  1 package not processed

Thanks
Magesh

EmilRyberg, 2020-10-21:
I also get this error on foxy, trying to build foxy-devel branch of this repo.

Magesh7803, 2020-10-22:
Upgrading pcl to the lastest version i.e. pcl-1.11 solved this issue.

closed 2020-10-23 by Magesh7803