ros-perception/perception_pcl #315: Multiple threads using toROSMsg causes node to crash

Zaaler, 2020-11-04:
Code referenced to is located in the perception_pcl.h

I've been using toROSMsg to convert PCLPointClouds<T> to sensor_msgs::msg::PointCloud2 and ran into a problem. If I pass in a sensor_msgs::msg::PointCloud2::SharedPtr, the node will crash on the assignment of:
auto time_stamp = rclcpp::Time(header.stamp);

So, I switched to using just a sensor_msgs::msg::PointCloud2 and the clouds were populated but, the timestamp fails to get set from the PCL Point Cloud timestamp. The function that is supposed to set the timestamp is:
 inline
  void fromPCL(const pcl::PCLHeader &pcl_header, std_msgs::msg::Header &header)
  {
    auto time_stamp = rclcpp::Time(header.stamp);
    fromPCL(pcl_header.stamp, time_stamp);
    header.frame_id = pcl_header.frame_id;
  }

I don't think the time_stamp variable used to hold the pcl_header.stamp value is ever set back to the header.stamp field.
The code would work with... I believe:
fromPCL(pcl_header.stamp, header.stamp);

I can wait for the toROSMsg function call to return and then I am able to correct the issue by manually setting pclpointcloud timestamp to the sensor_msgs timestamp using the equation in the perception_pcl header file:
stamp = rclcpp::Time(pcl_stamp * 1000ull); // Convert from us to ns

Not sure if this is a threading/protected memory problem but, I can only get the function to work with:
sensor_msgs::msg::PointCloud2 objects and then manually filling in the timestamp.

mvieth, 2020-11-27:
Perhaps you should post a larger snippet of your code (where you call the convert function). When you use the shared pointer, do you initialize it? (e.g. with `reset`)
Possibly important: what ROS version do you use (I see that you use ROS 2, but which one specifically?), what version of perception_pcl, how did you install it (compiled from source, apt, something else)?

SteveMacenski, 2020-12-04:
PR has been merged

closed 2020-12-04 by SteveMacenski