ros-perception/perception_pcl #318: Ubuntu xenial error libvtkproj4-6.2.so.6.2.0

ivan1993br, 2020-12-01:
Hi guys,

## Problem
When compiling a 2d costmap layer I had this problem `No rule to make target '/usr/lib/x86_64-linux-gnu/libvtkproj4-6.2.so.6.2.0', needed by '/home/mccr/dev/devel/lib/libcostmap_sonar_layer.so'`.

## Possible cause
 I was tracking the problem and I think it is in the dependency of this package the `libpcl-dev` that depends on `libvtk6.2-dev` that on xenial it is a osrf release `6.2.0+dfsg1-10build1+debian11.1+osrf1`, however on 19-11-2020 there was a new release of this package on ubuntu `6.2.0+dfsg1-10ubuntu0.1 `, that has priority on installation along with its dependencies.

## Solution
I think that a new release is needed.

Is anyone having this problem?

fmessmer, 2020-12-01:
yes, I see the same issue

I came across it during my kinetic CI jobs for packages that depend on `costmap_2d`:
https://github.com/ros-planning/navigation/issues/1058

SteveMacenski, 2020-12-01:
Ditto to Ferg's comment:

```
So we don't specify that directory in our CMake - which means that some upstream project (likely PCL) is exporting that as an include directory, but failing to properly specify the dependencies which make that directory exist.

If you install "libhdf5-mpi-dev libhdf5-openmpi-10 libhdf5-openmpi-dev" does it build then? If so, we could add those extra dependencies to hack around the PCL breakage (although it would be better to fix this upstream, that could take a lot longer).
```

We haven't changed anything here or re-released

ivan1993br, 2020-12-01:
I do not know if I got your point, but yes, the problem is not exactly on this package per say, but on its dependencies.

What I think that happened is that: pcl-ros dependes on libpcl, and libpcl depends on libvtk6, so when it was released with OSRF packages, it used the libvtk6 released by them(`6.2.0+dfsg1-10build1+debian11.1+osrf1`), which was fine, because when we include their repos, they were the most update one, now they are not, as ubuntu official released a new one. 

So, all the libs needed with on the (pcl-ros)Config.cmake, created with catkin-package, asks for the libs present on the OSRF custom libvtk6, that are not the same of the official one. So, I see, there is nothing in addition that we can install that can fix this, even adding the libhdf5, that does not solve the problem, you would get `libvtkproj4-6.2.so.6.2.0`. We would need to replace all the changes made by OSRF on the `libvbtk6`, I do not know if this can be easily done.

What I though as solution was that OSRF released a new version of libvtk6 with the latest available on ubutu xenial (the modification was minimal, [just dependency stuff](http://launchpadlibrarian.net/507795144/vtk6_6.2.0+dfsg1-10build1_6.2.0+dfsg1-10ubuntu0.1.diff.gz) ).

mhwasil, 2020-12-01:
I got the same issue just recently

SteveMacenski, 2020-12-01:
I think what Fergs and I mean is that this isn't an issue for this particular project. I recognize that its definitely a problem overall, but I think there's a different repo / group that should be looped in to help resolve it (since this isn't a `perception_pcl` or `navigation` codebase issue).

Can you identify a specific vtk release that broke this? If so, can you file a ticket / email a maintainer that was responsible for that release? While filing it here is good for visibility, Nav/PCL Perception maintainers are not the right persons to fix it :wink:

kyrofa, 2020-12-02:
I think @ivan1993br is right. The ubuntu update was fixing [this bug](https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1573234). We were in a weird spot, where vtk6 was building against the system libproj, but pcl had vtk's vendored libproj written into its cmake config files which caused compile errors. To fix this, OR backported a vtk6 fix from Debian that reverted back to using the vtk-vendored libproj, but that would have broken ABI in Xenial so we had to go the opposite direction to fix it. Now ROS deps are wanting the vendored libproj when Ubuntu uses the system libproj.

An easy solution would be for OR to continue carrying their patch and bump versions to stay on top of Ubuntu's, but I'm sure they would agree it would be nice not to have to do that.

mintar, 2020-12-03:
For people coming here just looking for a quick fix while waiting for the new release: Simply clone this repo (perception_pcl) into your Catkin workspace to build from source.

ivan1993br, 2020-12-03:
>For people coming here just looking for a quick fix while waiting for the new release: Simply clone this repo (perception_pcl) into your Catkin workspace to build from source.

Just an alternative: Pin the version on apt, editing apt preferences:
```text
Package: *
Pin: origin packages.ros.org
Pin-Priority: 1000
```
See [here](https://wiki.debian.org/AptConfiguration) for more information, this will use the libs released by osrf, this config will need to be removed later.

mintar, 2020-12-03:
That's also a great solution! Minor note: The file should be called `/etc/apt/preferences.d/ros-pin`.

jarvisschultz, 2020-12-03:
One other quick note. If you do have the `vtk` packages from OSRF installed you may also need to manually install `libproj-dev` since the OSRF versions don't have a dependency on that package. If that package is missing you may see errors like:

```
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj.so', needed by '/home/user/workspace/<TARGET>'.  Stop.
```

ivan1993br, 2020-12-03:
>One other quick note. If you do have the vtk packages from OSRF installed you may also need to manually install libproj-dev since the OSRF versions don't have a dependency on that package. If that package is missing you may see errors like:

This error is with vtk and libpcl from OSRF? Curious, because I think that the pcl from OSRF does not use system libproj, but instead vtk one.

jarvisschultz, 2020-12-03:
Well, `vtk` is from OSRF, but `libpcl1.7` is provided by packages on Ubuntu's repositories. When I see this error:
```
jarvis@jschultz:~/..2../src/nodelet_pcl_demo [tmp]⟫ dpkg -l |grep "libpcl1.7\|perception-pcl"
ii  libpcl1.7                                                   1.7.2-14build1.1                                            amd64        Point Cloud Library -- metapackage
ii  ros-kinetic-perception-pcl                                  1.4.4-0xenial-20201103-090055+0000                          amd64        PCL (Point Cloud Library) ROS interface stack.

jarvis@jschultz:~/..2../src/nodelet_pcl_demo [tmp]⟫ dpkg -l |grep "vtk"
hi  libvtk6-dev                                                 6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK header files
ii  libvtk6-java                                                6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Visualization Toolkit - A high level 3D visualization library - java
ii  libvtk6-qt-dev                                              6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK header files, containing Qt files
ii  libvtk6.2                                                   6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK libraries
ii  libvtk6.2-qt                                                6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK libraries, Qt files
ii  python-vtk6                                                 6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Python bindings for VTK
ii  tcl-vtk6                                                    6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Tcl bindings for VTK
ii  vtk6                                                        6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Binaries for VTK6

jarvis@jschultz:~/..2../src/nodelet_pcl_demo [tmp]⟫ apt-cache policy libpcl1.7
libpcl1.7:
  Installed: 1.7.2-14build1.1
  Candidate: 1.7.2-14build1.1
  Version table:
 *** 1.7.2-14build1.1 500
        500 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages
        100 /var/lib/dpkg/status
     1.7.2-14build1 500
        500 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
        500 https://dl.cloudsmith.io/Js9gREpP5F8aJSrj/outrider/truck/deb/ubuntu xenial/main amd64 Packages
```

So I think that the the issue is that there was a [new release of PCL](https://launchpad.net/ubuntu/+source/pcl/1.7.2-14build1.1) that went along with the updates to VTK.

jarvisschultz, 2020-12-03:
Well, `vtk` is from OSRF, but `libpcl1.7` is provided by packages on Ubuntu's repositories. When I see this error:
```
jarvis@jschultz:~/..2../src/nodelet_pcl_demo [tmp]⟫ dpkg -l |grep "libpcl1.7\|perception-pcl"
ii  libpcl1.7                                                   1.7.2-14build1.1                                            amd64        Point Cloud Library -- metapackage
ii  ros-kinetic-perception-pcl                                  1.4.4-0xenial-20201103-090055+0000                          amd64        PCL (Point Cloud Library) ROS interface stack.

jarvis@jschultz:~/..2../src/nodelet_pcl_demo [tmp]⟫ dpkg -l |grep "vtk"
hi  libvtk6-dev                                                 6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK header files
ii  libvtk6-java                                                6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Visualization Toolkit - A high level 3D visualization library - java
ii  libvtk6-qt-dev                                              6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK header files, containing Qt files
ii  libvtk6.2                                                   6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK libraries
ii  libvtk6.2-qt                                                6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        VTK libraries, Qt files
ii  python-vtk6                                                 6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Python bindings for VTK
ii  tcl-vtk6                                                    6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Tcl bindings for VTK
ii  vtk6                                                        6.2.0+dfsg1-10build1+debian11.1+osrf1                       amd64        Binaries for VTK6

jarvis@jschultz:~/..2../src/nodelet_pcl_demo [tmp]⟫ apt-cache policy libpcl1.7
libpcl1.7:
  Installed: 1.7.2-14build1.1
  Candidate: 1.7.2-14build1.1
  Version table:
 *** 1.7.2-14build1.1 500
        500 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages
        100 /var/lib/dpkg/status
     1.7.2-14build1 500
        500 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
```

So I think that the the issue is that there was a [new release of PCL](https://launchpad.net/ubuntu/+source/pcl/1.7.2-14build1.1) that went along with the updates to VTK.

kyrofa, 2020-12-03:
> So I think that the the issue is that there was a new release of PCL that went along with the updates to VTK.

Indeed, [this post](https://github.com/ros/rosdistro/issues/27538#issuecomment-737498157) hopefully breaks thing down in a way that makes sense.

Hexonberg, 2020-12-06:
try this 

sudo ln -s /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so /usr/lib/x86_64-linux-gnu/libvtkproj4-6.2.so.6.2.0

ALTinners, 2020-12-07:
For those wanting a pinning solution like [this one](https://github.com/ros-perception/perception_pcl/issues/318#issuecomment-737942173) but less overarching - try this 

`/etc/apt/preferences.d/99-ros-pin-for-pcl`

```
Package: libvtk6-dev libvtk6.2 libvtk6-java python-vtk6 tcl-vtk6 vtk6 libvtk6.2-qt libvtk6-qt-dev 
Pin: version 6.2.0+dfsg1-10ubuntu0.1
Pin-Priority: -1

Package: libvtk6-dev libvtk6.2 libvtk6-java python-vtk6 tcl-vtk6 vtk6 libvtk6.2-qt libvtk6-qt-dev 
Pin: version 6.2.0+dfsg1-10build1+debian11.1+osrf1
Pin-Priority: 600
```

Strictly speaking I don't think the `-1` pin is necessary - but I've never done pinning before so ... yeah.

ivan1993br, 2020-12-07:
Ditto to tfoot comment:
>The sync is out: https://discourse.ros.org/t/new-packages-for-kinetic-kame-2020-12-07/17786

closed 2020-12-07 by ivan1993br