ros-perception/perception_pcl #320: pcl/conversions.h: No such file or directory  #include <pcl/conversions.h>           ^~~~~~~~~~~~~~~~~~~

Schloern93, 2020-12-24:
Hello

Ros distro: eloquent on Ubuntu 18.04

I have cloned the package for Ros eloquent into my worspace. I was able to build the package successfully.
Now when I try to include the package I get the following error:
/opt/ros/eloquent/include/pcl_conversions/pcl_conversions.h:46: error: pcl/conversions.h: No such file or directory
 #include <pcl/conversions.h>

I am still quite new to ros. Therefore the perhaps a rather stupid question. How can I include the package in my project after successful build?
Do I need to install this into my ros library first? If so how exactly do I have to do that

package.xml

<depend>pcl_conversions</depend>

CMakeList.txt

find_package(pcl_conversions REQUIRED)
add_executable(object_detection_node src/object_detection_node.cpp src/detection_model.cpp src/visualisation_rviz.cpp)
ament_target_dependencies(object_detection_node rclcpp std_msgs sensor_msgs visualization_msgs nav_msgs tf2 darknet_ros_msgs pcl_conversions)

Thanks for the help in advance.

mvieth, 2020-12-24:
Hi @Schloern93 ! `pcl/conversions.h` is a header from PCL itself, so most obvious question first: is PCL installed? Are the PCL header files in the cmake search path?

Schloern93, 2020-12-24:
Thanks for the quick reply
I have installed pcl_conversion with the following command
sudo apt install ros-eloquent-pcl-conversions

The CMake path should be set by the following command in CMakeLst.txt or am I missing something?
find_package(pcl_conversions REQUIRED)

mvieth, 2020-12-24:
I was talking about PCL (the Point Cloud Library https://github.com/PointCloudLibrary/pcl). `pcl-conversions` is the package that handles conversions between PCL and ROS. But if you installed `ros-eloquent-pcl-conversions` via apt, then PCL should be automatically installed.
I am not 100% sure if there is something specific about this in ROS2, but you would usually have to do something like this: `include_directories(PCL_CONVERSIONS_INCLUDE_DIRS)` (not sure about the name of the variable).

Schloern93, 2020-12-24:
Thanks for the help! 
I forgot the following line in my CmakeList.txt
target_link_libraries(object_detection_node ${PCL_LIBRARIES})

I have one last question now. I want to evaluate each PointCloud2 msgs and get the points XYZ.
Does this already work for Ros2?

mvieth, 2020-12-24:
> I have one last question now. I want to evaluate each PointCloud2 msgs and get the points XYZ.
Does this already work for Ros2?

Functionality-wise, the ROS1 and ROS2 versions of this repo should be equivalent, as far as I know.

Schloern93, 2020-12-24:
I have now tried after building the package to include this in my package in the same workspace. 
Unfortunately, I still can't do it.

My project builds without errors.
But as soon as I try to insert pcl_ros into my .cpp file or to do this in my CMakeList.txt with 
find_package(pcl_ros REQUIRED)
I get the following compiler error:

CMakeLists.txt:29: error: By not providing "Findpcl_ros.cmake" in CMAKE_MODULE_PATH this project has asked CMake to find a package configuration file provided by "pcl_ros", but CMake did not find one. Could not find a package configuration file provided by "pcl_ros" with any of the following names: pcl_rosConfig.cmake pcl_ros-config.cmake Add the installation prefix of "pcl_ros" to CMAKE_PREFIX_PATH or set "pcl_ros_DIR" to a directory containing one of the above files.  If "pcl_ros" provides a separate development package or SDK, be sure it has been installed.

So my question: do I need to build this package in a separate workspace and then add it to the ptc library with an installation command? 
I don't have much experience with CMake yet, so this might be a stupid question.

Thanks for the help

mvieth, 2020-12-24:
Did you install `ros-eloquent-pcl-ros`?
I am not sure what the recommended way is to do this in ROS2, you might have to wait for someone who knows. In ROS1 it is usually done like this: `find_package(catkin COMPONENTS pcl_ros)`

SteveMacenski, 2020-12-24:
Pcl ros is not yet ported to ROS2.

Schloern93, 2020-12-25:
Thanks for the quick info.
Is there a way to simply edit the PointCloud2 data by hand to get the necessary coordinates of the pixels?

mvieth, 2020-12-25:
> Pcl ros is not yet ported to ROS2.

But pcl_conversions is already ported, right? So converting a PointCloud2 message to a `pcl::PointCloud<pcl::PointXYZ>` for example should work?

SteveMacenski, 2020-12-28:
Yes. PCL conversions is released and ready to rock and roll

Schloern93, 2020-12-29:
How exactly can the conversion of the points be done?

I don't understand how to use the functions. 
I have the following problem:
1. I use an Intel Realsense D435 camera. From this camera I receive a PointCloud2 message.

Used pcl includes
#include <pcl/conversions.h>
#include <pcl/point_types.h>
#include <pcl/point_cloud.h>
#include <pcl/PCLPointCloud2.h>
#include <pcl/PCLPointField.h>

Here is my Ros2 callback:
void callback_depth_image_d435 (const sensor_msgs::msg::PointCloud2::SharedPtr point_cloud2_msgs)
    {
      printf ("Cloud: width = %d, height = %d\n", point_cloud2_msgs->width, point_cloud2_msgs->height);

      pcl::PointCloud<pcl::PointXYZ> point_cloud;

    }

2. i am now trying to access the points as in the example linked below. (point_cloud2_msgs->points) But I can only access the whole data matrix (point_cloud2_msgs->data). 
Can anyone explain to me how I can access the individual points using the pcl/conversion? (Maybe a short code example)

http://wiki.ros.org/pcl_ros

Thanks for the help

mvieth, 2020-12-29:
`pcl::fromROSMsg(point_cloud2_msgs, point_cloud)` [for API see here](https://docs.ros.org/en/noetic/api/pcl_conversions/html/namespacepcl.html#af662c7d46db4cf6f7cfdc2aaf4439760)

Alternatively you could try to change the callback to `void callback(const pcl::PointCloud<pcl::PointXYZ>::ConstPtr& msg)`, then the conversion should happen implicitly.

Schloern93, 2020-12-29:
Thanks to all for the help. It has worked for me. 
This is my function call:

     void callback_depth_image_d435 (const sensor_msgs::msg::PointCloud2::SharedPtr point_cloud2_msgs)
     {
      
      pcl::PointCloud<pcl::PointXYZ> point_cloud;
      pcl::fromROSMsg(*point_cloud2_msgs, point_cloud);     

      BOOST_FOREACH(const pcl::PointXYZ& pt, point_cloud.points)
      {          
          std::cout  << "x: " << pt.x <<"\n";
          std::cout  << "y: " << pt.y <<"\n";
          std::cout  << "z: " << pt.z <<"\n";
          std::cout << "---------" << "\n";
      }

closed 2020-12-29 by Schloern93