ros-perception/perception_pcl #321: AUR installation fails due to outdated include - ROS Noetic

TheCodeSummoner, 2020-12-25:
### Issue

The package can't be installed from AUR.

The following console output showcases the issue:

```
CMake Error at /opt/ros/noetic/share/pcl_conversions/cmake/pcl_conversionsConfig.cmake:113 (message):
  Project 'pcl_conversions' specifies '/usr/include/pcl-1.10' as an include
  dir, which is not found.  It does neither exist as an absolute directory
  nor in '${prefix}//usr/include/pcl-1.10'.  Check the issue tracker
  'https://github.com/ros-perception/perception_pcl/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:39 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/florian/.cache/yay/ros-noetic-pcl-ros/src/build/CMakeFiles/CMakeOutput.log".
See also "/home/florian/.cache/yay/ros-noetic-pcl-ros/src/build/CMakeFiles/CMakeError.log".
==> ERROR: A failure occurred in build().
    Aborting...
error making: ros-noetic-pcl-ros
```

Running `ls /usr/include | grep pcl` yields:

```
pcl-1.11
```

The solution would be to ask for `pcl-1.11` instead of `pcl-1.10`.

### Environment

#### OS

- Distributor ID:	ManjaroLinux
- Description:	Manjaro Linux
- Release:	20.2
- Codename:	Nibia

#### Package manager

- yay v10.1.0 - libalpm v12.0.2

TheCodeSummoner, 2020-12-25:
Actually, I just tried changing it manually in the config file and the build failed anyway:

```
[ 81%] Linking CXX shared library devel/lib/libpcl_ros_features.so
[ 81%] Built target pcl_ros_features
[ 82%] Linking CXX executable devel/lib/pcl_ros/bag_to_pcd
/usr/bin/ld: /opt/ros/noetic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/bag_to_pcd.dir/build.make:405: devel/lib/pcl_ros/bag_to_pcd] Error 1
make[1]: *** [CMakeFiles/Makefile2:2382: CMakeFiles/bag_to_pcd.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 84%] Linking CXX shared library devel/lib/libpcl_ros_filter.so
[ 84%] Built target pcl_ros_filter
[ 85%] Linking CXX shared library devel/lib/libpcl_ros_io.so
[ 85%] Built target pcl_ros_io
[ 87%] Linking CXX shared library devel/lib/libpcl_ros_segmentation.so
[ 87%] Built target pcl_ros_segmentation
make: *** [Makefile:149: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
error making: ros-noetic-pcl-ros
```

Is that a genuine error or something related to AUR version of the code being outdated?

SteveMacenski, 2020-12-28:
We do not support non-supported PCL versions. Each ROS distro has a specific supported PCL version. I’d be happy to field PRs to make it work with other PCL versions you may want to use, but maintainer cycles won’t be spent on it.

TheCodeSummoner, 2020-12-28:
Sorry, I'm new to using ROS and AUR in general, so things are still quite confusing to me. It seems that due to a recent update a lot of things starting falling over because of `libtf.so` (for example I can't even install ROS anymore), so I will investigate what's wrong at my package provider level instead. Thank you for your time.

closed 2020-12-28 by TheCodeSummoner