ros-perception/perception_pcl #322: macro names must be identifiers

ybw-yang, 2021-03-05:
--- stderr: pcl_ros                               
<command-line>:0:1: error: macro names must be identifiers

ybw-yang, 2021-03-05:
<command-line>:0:1: error: macro names must be identifiers
make[2]: *** [CMakeFiles/pcl_ros_tf.dir/src/transforms.cpp.o] Error 1
make[1]: *** [CMakeFiles/pcl_ros_tf.dir/all] Error 2
make: *** [all] Error 2
---
how to solve it?

parkdg44, 2021-06-26:
I don't Know why this is work,
But just change PCL to remark in CMakeLists.txt like this, it works in my case.

set(dependencies
  pcl_conversions
  rclcpp
  sensor_msgs
  geometry_msgs
  tf2
  tf2_geometry_msgs
  tf2_ros
  EIGEN3
  #PCL
)

ybw-yang, 2021-06-30:
It's useful, thanks very much!

> I don't Know why this is work,
> But just change PCL to remark in CMakeLists.txt like this, it works in my case.
> 
> set(dependencies
> pcl_conversions
> rclcpp
> sensor_msgs
> geometry_msgs
> tf2
> tf2_geometry_msgs
> tf2_ros
> EIGEN3
> #PCL ## Change This Line !!!
> )

closed 2021-06-30 by ybw-yang