ros-perception/perception_pcl #323: [Foxy] pcl_conversions build error: expected initializer before ‘<’ token

cardboardcode, 2021-03-10:
Hi, thank you for continuing to maintain this critical package in the ROS2 ecosystem. :smile: :+1:

# Issue Description
Unable to compile ROS2 packages that uses **pcl_conversions** in Docker :whale:  .

## Environment 
1. Ubuntu **20.04.1** LTS
2. **ros-foxy-pcl-conversions**/focal,now **2.2.0**-1focal.20210225.224542 amd64
3. **ros-foxy-pcl-msgs**/focal,now **1.0.0**-2focal.20210225.214430 amd64
4.  Docker version **19.03.8**, build afacb8b7f0

## Expected Behaviour
Compiles ROS2 packages that uses **pcl_conversions** without error.
```bash
Starting >>> cpp_pubsub
Finished <<< cpp_pubsub [5.76s]                     

Summary: 1 package finished [5.83s]
```

## Actual Behaviour
Compilation fails with the following errors:
```bash 
Starting >>> cpp_pubsub
--- stderr: cpp_pubsub                             
In file included from /opt/ros/foxy/include/pcl_msgs/msg/point_indices.hpp:9,
                 from /opt/ros/foxy/include/pcl_conversions/pcl_conversions.h:61,
                 from /home/test_usr/ros2_ws/src/cpp_pubsub/src/publisher_member_function.cpp:20:
/opt/ros/foxy/include/pcl_msgs/msg/detail/point_indices__traits.hpp:27:25: error: expected initializer before ‘<’ token
   27 | inline const char * name<pcl_msgs::msg::PointIndices>()
      |                         ^
In file included from /opt/ros/foxy/include/pcl_msgs/msg/model_coefficients.hpp:9,
                 from /opt/ros/foxy/include/pcl_conversions/pcl_conversions.h:64,
                 from /home/test_usr/ros2_ws/src/cpp_pubsub/src/publisher_member_function.cpp:20:
/opt/ros/foxy/include/pcl_msgs/msg/detail/model_coefficients__traits.hpp:27:25: error: expected initializer before ‘<’ token
   27 | inline const char * name<pcl_msgs::msg::ModelCoefficients>()
      |                         ^
In file included from /opt/ros/foxy/include/pcl_msgs/msg/vertices.hpp:9,
                 from /opt/ros/foxy/include/pcl_conversions/pcl_conversions.h:67,
                 from /home/test_usr/ros2_ws/src/cpp_pubsub/src/publisher_member_function.cpp:20:
/opt/ros/foxy/include/pcl_msgs/msg/detail/vertices__traits.hpp:23:25: error: expected initializer before ‘<’ token
   23 | inline const char * name<pcl_msgs::msg::Vertices>()
      |                         ^
In file included from /opt/ros/foxy/include/pcl_msgs/msg/polygon_mesh.hpp:9,
                 from /opt/ros/foxy/include/pcl_conversions/pcl_conversions.h:70,
                 from /home/test_usr/ros2_ws/src/cpp_pubsub/src/publisher_member_function.cpp:20:
/opt/ros/foxy/include/pcl_msgs/msg/detail/polygon_mesh__traits.hpp:29:25: error: expected initializer before ‘<’ token
   29 | inline const char * name<pcl_msgs::msg::PolygonMesh>()
      |                         ^
make[2]: *** [CMakeFiles/talker.dir/build.make:63: CMakeFiles/talker.dir/src/publisher_member_function.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/talker.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< cpp_pubsub [5.33s, exited with code 2]

Summary: 0 packages finished [5.48s]
  1 package failed: cpp_pubsub
  1 package had stderr output: cpp_pubsub
```

## Instructions to Reproduce Error
This section provides a minimal example to reproduce the compilation error shown above.
```bash
# Ensure that you have Docker installed.
docker pull ros:foxy-ros-base-focal

# Run the docker image as container.
docker run -it \
--name pcl_conversions_test \
-v /tmp/.X11-unix:/tmp/.X11-unix \
-e DISPLAY=$DISPLAY \
-u 0 \
ros:foxy-ros-base-focal /bin/bash
# You should now be in the docker container via a bash terminal.

# Install tools for editing
apt-get update && apt-get install wget nano -y

# Create example cpp_pubsub package
cd home
mkdir -p test_usr/dev_ws/src && cd test_usr/dev_ws/src
source /opt/ros/foxy/setup.bash
ros2 pkg create --build-type ament_cmake cpp_pubsub
cd cpp_pubsub/src
wget -O publisher_member_function.cpp https://raw.githubusercontent.com/ros2/examples/foxy/rclcpp/topics/minimal_publisher/member_function.cpp

# Install pcl_conversions in the docker container
export DEBIAN_FRONTEND="noninteractive"
apt-get install ros-foxy-pcl-conversions -y

# Include pcl_conversions in example cpp_pubsub package
# Replace the following package files as shown in the section, **Files To Replace**, below, 
nano /home/test_usr/dev_ws/src/cpp_pubsub/package.xml 
# Replace with content below.
nano /home/test_usr/dev_ws/src/cpp_pubsub/CMakeLists.txt 
# Replace with content below.
nano /home/test_usr/dev_ws/src/cpp_pubsub/src/publisher_member_function.cpp 
# Replace with content below.

# Build example package to generate above error.
cd /home/test_usr/dev_ws
colcon build
```

## Files To Replace
Please copy and replace the following content below into the designated **3 files**.
1. `/home/test_usr/dev_ws/src/cpp_pubsub/package.xml`
``` xml
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>cpp_pubsub</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="root@todo.todo">root</maintainer>
  <license>TODO: License declaration</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <depend>rclcpp</depend>
  <depend>std_msgs</depend>
  <depend>pcl_conversions</depend>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>

```
2. `/home/test_usr/dev_ws/src/cpp_pubsub/CMakeLists.txt`
```txt
cmake_minimum_required(VERSION 3.5)
project(cpp_pubsub)

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
find_package(PCL REQUIRED)
find_package(pcl_conversions REQUIRED)

include_directories(
  include
  ${PCL_INCLUDE_DIRS}
)

add_definitions(${PCL_DEFINITIONS})

add_executable(talker src/publisher_member_function.cpp)
ament_target_dependencies(talker rclcpp std_msgs pcl_conversions)
target_link_libraries(talker ${PCL_LIBRARIES})

install(TARGETS
  talker
  DESTINATION lib/${PROJECT_NAME})

ament_package()

```
3. `/home/test_usr/dev_ws/src/cpp_pubsub/src/publisher_member_function.cpp`
``` cpp
#include <chrono>
#include <memory>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"
#include "pcl_conversions/pcl_conversions.h"

using namespace std::chrono_literals;

/* This example creates a subclass of Node and uses std::bind() to register a
 * member function as a callback from the timer. */

class MinimalPublisher : public rclcpp::Node
{
public:
  MinimalPublisher()
  : Node("minimal_publisher"), count_(0)
  {
    publisher_ = this->create_publisher<std_msgs::msg::String>("topic", 10);
    timer_ = this->create_wall_timer(
      500ms, std::bind(&MinimalPublisher::timer_callback, this));
  }

private:
  void timer_callback()
  {
    auto message = std_msgs::msg::String();
    message.data = "Hello, world! " + std::to_string(count_++);
    RCLCPP_INFO(this->get_logger(), "Publishing: '%s'", message.data.c_str());
    publisher_->publish(message);
  }
  rclcpp::TimerBase::SharedPtr timer_;
  rclcpp::Publisher<std_msgs::msg::String>::SharedPtr publisher_;
  size_t count_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalPublisher>());
  rclcpp::shutdown();
  return 0;
}
```
## Other Info
1. Verified that **pcl_conversions** can be compiled successfully outside of Docker on native Ubuntu 20.04.
2. Verified that both **ros-foxy-pcl-conversions** and **ros-foxy-pcl-msgs** are of the same version within and outside of Docker.
3. Verified that **C++14** is being used to compile within and outside of Docker.
4. Verified that **libpcl-dev** is of the same version within and outside of Docker. 
5. Verified that **gcc** version differences does not contribute to the error. 
6. The reason for trying to build **pcl_conversions** in a Docker is to provide Continuous Integration (CI) builds for ROS2 package  that handle PointClouds data.

SteveMacenski, 2021-05-20:
A PR to resolve would be appreciated

cardboardcode, 2021-05-21:
After attempting to reproduce the error described, I can no longer reproduce it. 

## Results
```bash
root@86ee85153c9f:/home/test_usr/dev_ws# colcon build
Starting >>> cpp_pubsub
--- stderr: cpp_pubsub                              
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to openni will be disabled
** WARNING ** visualization features related to openni2 will be disabled
** WARNING ** apps features related to openni will be disabled
---
Finished <<< cpp_pubsub [8.68s]

Summary: 1 package finished [8.81s]
  1 package had stderr output: cpp_pubsub

```
This issue must have been resolved by the recent commits. Not sure which though. 

Anyway, closing issue.

closed 2021-05-21 by cardboardcode