ros-perception/perception_pcl #324: New release to Rolling?

wjwwood, 2021-03-18:
@mvieth @SteveMacenski we're preparing to fork rolling into galactic and we noticed there are several changes here on 'foxy-devel' that have not been released, https://github.com/ros-perception/perception_pcl/compare/2.2.0...foxy-devel.

Does anyone have time to do a release? Or if the changes are important, but you don't have time let us know and we can do a release on your behalf.

mvieth, 2021-03-19:
Hi! I am actually not a maintainer for perception_pcl, so I do not have the permissions, but IMO a release would be great :smiley:

SteveMacenski, 2021-03-19:
I think we'd happy to give you maintainer's permissions if you wanted to help maintain this project :wink: 

PCL ROS is still [largely unported](https://github.com/ros-perception/perception_pcl/blob/foxy-devel/pcl_ros/CMakeLists.txt) so I don't think it makes sense to release those changes into Foxy.

mvieth, 2021-03-22:
> I think we'd happy to give you maintainer's permissions if you wanted to help maintain this project

Thank you for the offer, but I'm already a maintainer of PCL, which takes up quite some time. I don't think I also have the time to maintain perception_pcl currently. I might work on issues from time to time, though.

SteveMacenski, 2021-05-20:
In progress

closed 2021-05-20 by SteveMacenski