ros-perception/perception_pcl #325: [compile error] "PCL conversions dont compile"

lidiaxp, 2021-04-27:
I trying to use a pcl with statistical filter and ROS.

I am using this [code](https://drive.google.com/file/d/1MWKKwsLDyzNMkdhi73HyXj60OHucvMnI/view?usp=sharing) with this [cmakeList](https://drive.google.com/file/d/1xmn-zw6nHZLdYBOoB29odQaZHYsnwSGr/view?usp=sharing).

When I try compile appear this error:

`collect2: error: ld returned 1 exit status
make[2]: *** [pcl_manipulation/CMakeFiles/pcl_sample_visualizer.dir/build.make:409: /home/lidiaxp/workspace/devel/lib/pcl_manipulation/pcl_sample_visualizer] Error 1
make[1]: *** [CMakeFiles/Makefile2:6445: pcl_manipulation/CMakeFiles/pcl_sample_visualizer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
`

`collect2: error: ld returned 1 exit status
make[2]: *** [pcl_manipulation/CMakeFiles/pcl_sample_filter.dir/build.make:409: /home/lidiaxp/workspace/devel/lib/pcl_manipulation/pcl_sample_filter] Error 1
make[1]: *** [CMakeFiles/Makefile2:6499: pcl_manipulation/CMakeFiles/pcl_sample_filter.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
`

I believe that this error is because the pcl conversions package because when I try run a package without it works. 

I install this package and ros-noetic-pcl-conversions. I am using Ubuntu 20.04 with ROS noetic and Python 3.8.

Ps. When I tested this code in another computer it works. So, I guess that the problem is some library that I have installed, as ros-noetic-pcl-conversions, or [this](https://github.com/PointCloudLibrary/pcl) (PCL Library), and I know don't how uninstall.

SteveMacenski, 2021-04-27:
Ask on ROS Answers, this is not a bug but probably not linking correctly.

closed 2021-04-27 by SteveMacenski

lidiaxp, 2021-04-27:
ok, so I asked [here](https://answers.ros.org/question/377161/compile-error-pcl-conversions-dont-compile/)