ros-perception/perception_pcl #328: Timestamp during pcl::PointCloud to sensor_msgs::msg::PointCloud2 conversion is not assigned

zygfrydw, 2021-06-01:
**branch: foxy-devel**

The function `void toROSMsg(const pcl::PointCloud<T> &pcl_cloud, sensor_msgs::msg::PointCloud2 &cloud)` converts type pcl::PointCloud to sensor_msgs::msg::PointCloud2 and it uses the following code to do the header conversion
```cpp
 inline void fromPCL(const pcl::PCLHeader &pcl_header, std_msgs::msg::Header &header)
  {
    auto time_stamp = rclcpp::Time(header.stamp);
    fromPCL(pcl_header.stamp, time_stamp);
    header.frame_id = pcl_header.frame_id;
  }
```
This code does not assign the timestamp to `header` object, leaving it empty. 

The correct implementation should be:
```cpp
 inline void fromPCL(const pcl::PCLHeader &pcl_header, std_msgs::msg::Header &header)
  {
    rclcpp::Time time_stamp;
    fromPCL(pcl_header.stamp, time_stamp);
    header.frame_id = pcl_header.frame_id;
    header.stamp = time_stamp;
  }
```

mvieth, 2021-06-01:
It seems you are not looking at the most recent version of the branch:

https://github.com/ros-perception/perception_pcl/blob/foxy-devel/pcl_conversions/include/pcl_conversions/pcl_conversions.h#L111-L116

This commit has fixed that:
https://github.com/ros-perception/perception_pcl/commit/135a2c29e3f46904f586e963cc4a4ec3fe594a9c#diff-b94bf92207242115a56ee619ba274a14243f3aae7cd8e0e443afb3ea72db8b3a

zygfrydw, 2021-06-14:
I was checking source codes for perception_pcl 2.2.0 because it is the version available on the official apt server for ros-foxy. 
When the version 2.3.1 with the aforementioned modifications will be available on the official apt server?

zygfrydw, 2021-06-14:
I was checking the commit history and I have found that the package version was changed to 2.3.1 on galactic and ros2 branches, while the version was not changed for a foxy branch.  
Is it possible that the package in the apt repository is built from 2.2.0 tag or that the new package was not deployed due to republishing package with the same version number?

SteveMacenski, 2021-09-08:
Sync coming next week

closed 2021-09-08 by SteveMacenski