ros-perception/perception_pcl #333: Undefined behavior in point_cloud.h

richmattes, 2021-06-30:
The point cloud serialization code in `point_cloud.h` relies on undefined behavior when using `operator[]()` to access element 0 of a vector when the vector is empty, as implemented here (and potentially elsewhere):
https://github.com/ros-perception/perception_pcl/blob/bd21be2cb8ec6ad70a282c61b14aa9dd43edf9ff/pcl_ros/include/pcl_ros/point_cloud.h#L167

C++11 [added a data()](http://www.open-std.org/jtc1/sc22/wg21/docs/lwg-defects.html#464) method to `std::vector` to avoid this issue - `point_cloud.h` should be updated to use `.data()` instead of a reference to the first vector element to get data pointers from std::vectors.  Alternately, a size check should be performed before accessing `vector[0]`.

SteveMacenski, 2021-09-14:
Sounds good - can you submit a PR?

closed 2021-09-27 by SteveMacenski

richmattes, 2022-02-05:
Any chance you can do a new release for melodic and noetic in the near-term?  It looks like there have been a couple other PRs/fixes since 1.7.3

SteveMacenski, 2022-02-05:
Sure thing, can't do at this moment but I'll mark this on my queue, hopefully next week.

PR #346 (2021-09-27): pcl_ros: Use vec.data() instead of &vec[0]
Update pcl_ros to use the data() member function of STL containers to
get the pointer to the underlying storage, instead of dereferencing the
zeroth element and taking its reference.  When a container is of size 0,
dereferencing element 0 with operator[]() is undefined behavior, and
will trigger assertions when features like _GLIBCXX_ASSERTIONS are
enabled.

Fixes #333.
commit 8969ea63 (2021-09-27): pcl_ros: Use vec.data() instead of &vec[0] (#346)

* pcl_ros: Use vec.data() instead of &vec[0]

Update pcl_ros to use the data() member function of STL containers to
get the pointer to the underlying storage, instead of dereferencing the
zeroth element and taking its reference.  When a container is of size 0,
dereferencing element 0 with operator[]() is undefined behavior, and
will trigger assertions when features like _GLIBCXX_ASSERTIONS are
enabled.

Fixes #333.

* Use CMake to manage C++ standard when possible

CMake 3.1 introduced the CMAKE_CXX_STANDARD to set C++ standard versions
used b [... cut]
commit 755f5663 (2021-09-27): pcl_ros: Use vec.data() instead of &vec[0] (#348)

Update pcl_ros to use the data() member function of STL containers to
get the pointer to the underlying storage, instead of dereferencing the
zeroth element and taking its reference.  When a container is of size 0,
dereferencing element 0 with operator[]() is undefined behavior, and
will trigger assertions when features like _GLIBCXX_ASSERTIONS are
enabled.

Fixes #333.