ros-perception/perception_pcl #336: Build error for Galactic on Ubuntu 18

karl-schulz, 2021-07-26:
Hi,
when building pcl_ros on Ubuntu 18 (L4T, any NVIDIA Jetson), I get the following error:

```
<command-line>:0:1: error: macro names must be identifiers
make[2]: *** [CMakeFiles/pcl_ros_tf.dir/src/transforms.cpp.o] Error 1
make[1]: *** [CMakeFiles/pcl_ros_tf.dir/all] Error 2
make: *** [all] Error 2
```

System:

* Ubuntu 18
* ROS2 Galactic
* `pcl_ros` branch `galactic` or `ros2`

The issue seems to be the `PCL` in `ament_add_dependencies` (line 44). There is not much more error output and I have difficulties troubleshooting this. There seems to be a preprocessor instruction to be failing. The package builds when `src/transforms.cpp` is an empty file and I remove `PCL_ROS` from the dependencies. As soon as I add `PCL_ROS` (event with empty `transforms.cpp`) the build fails.

Is this an issue with the CMakeLists on bionic? Building on Ubuntu 20 works. Is the build supposed to work on Ubuntu 18? 

(I know it is officially not supported, but the NVIDIA Jetsons with a large user base in robotics are only running with Ubuntu 18)

SteveMacenski, 2021-07-26:
I can't say I know what that is off hand, that's probably having to do with the PCL versions being incompatible, if I were to guess. It could be CMake versions too I suppose. 

This isn't something I can easily help with from afar, unfortunately. This is the kind of bug I'd have to spend a few days digging into if it were in my systems and I could poke it / change things until I made some progress. Happy to help answer questions to find a solution and merge in a patch if possible. 

Googling this issue though comes up with a few good resources https://github.com/ros-industrial/yak/issues/51 https://stackoverflow.com/questions/10779872/command-line11-error-macro-names-must-be-identifiers, even within the ROS / PCL communities

karl-schulz, 2021-07-27:
Thanks for the input. I indicated the package versions in the first post and will try building PCL=1.10 from source and see if theis resolves the problem!

karl-schulz, 2021-07-27:
Great, building pcl=1.10 from source resolved the issue.

For anyone having the same issue:
* Purge pcl <1.10 from your machine  
  `sudo apt remove libpcl-* --purge`
* Download PCL 1.10, build and install it from source  
  See https://pcl.readthedocs.io/projects/tutorials/en/latest/compiling_pcl_posix.html
* Remove your build cache to force cmake to find the new PCL  
  `rm -rf ros_ws/build*`
* Re-Build the workspace

closed 2021-07-27 by karl-schulz