ros-perception/perception_pcl #337: Installing perception_pcl from source

DanielFreer-antobot, 2021-07-27:
Hi, I am attempting to use a sort of strange configuration for my system due to factors unrelated to PCL. In the end, I am using ROS Noetic on Ubuntu 18.04, which means that most ROS packages are not available for installation through apt. 

Could you provide some guidance on how to install perception_pcl (and the other two packages within this repo) from source?

When I clone the source code into catkin_ws/src/ and attempt to use catkin_make, all three of these packages (pcl_conversions, pcl_ros, and perception_pcl) are listed as "plain cmake" packages, so the build stops. Another package I'm using (rtabmap_ros) is dependent on these 3 packages, so it seems necessary to install them from source within the catkin workspace. Also, when I look at the packages, they seem like ROS packages, so I'm not sure what the issue is.

If you have any suggestions or advice for achieving this, please let me know.

SteveMacenski, 2021-07-27:
Please ask on ROS Answers, this issue tracker isn't for specific advice outside of the distribution's officially supported software versions / OS's.

closed 2021-07-27 by SteveMacenski

DanielFreer-antobot, 2021-07-27:
Hi Steve,

I think I just figured out the issue anyway - I was on the ROS2 branch when I was trying to build. Now that I have gotten onto ROS Melodic, it seems to be working (for now).

Thanks anyway!

Petros626, 2022-04-22:
@DanielFreer-antobot  can you explain how to install this package on ROS1 noetic pls?