ros-perception/perception_pcl #345: Build failing with Ubuntu 20.04 and ROS Noetic

Enes1097, 2021-09-21:
Hey,
I just discovered this package and was about to use it, but i stumbled upon some problems.
In this case I tried it with the melodic and ROS2 branch on this repository, but got an error message while building the workspace with "catkin_make":

'ros_pcl_ws$ catkin_make
Base path: /home/enescelik/ros_pcl_ws
Source space: /home/enescelik/ros_pcl_ws/src
Build space: /home/enescelik/ros_pcl_ws/build
Devel space: /home/enescelik/ros_pcl_ws/devel
Install space: /home/enescelik/ros_pcl_ws/install
####
#### Running command: "cmake /home/enescelik/ros_pcl_ws/src -DCATKIN_DEVEL_PREFIX=/home/enescelik/ros_pcl_ws/devel -DCMAKE_INSTALL_PREFIX=/home/enescelik/ros_pcl_ws/install -G Unix Makefiles" in "/home/enescelik/ros_pcl_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/enescelik/ros_pcl_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/enescelik/ros_pcl_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - perception_pcl (plain cmake)
-- ~~  - pcl_conversions (plain cmake)
-- ~~  - pcl_ros (plain cmake)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/enescelik/ros_pcl_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/enescelik/ros_pcl_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
`
I also tried the kinetic branch on a Virtual Machine with Ubuntu 18.04 and the same error messsage appeared.

Help would be much appreciated :)

Cheers

thschafer, 2021-11-10:
I just built the melodic-devel branch successfully. I am on Ubuntu 20.04 running ROS1 Noetic. 
Your build is not actually failing; CMake is quitting at the configuration stage, so it never attempts a build. 
The CMake Error is indicating that the issue is related to a non-catkin package in your workspace. I suspect you have other packages besides pcl which are causing this problem. Have you tried 'catkin_make_isolated', as suggested by the error message?

tomlankhorst, 2021-11-13:
The error message literally says:

> This workspace contains non-catkin packages in it, and catkin cannot build a non-homogeneous workspace without isolation. Try the `catkin_make_isolated` command instead.

So use `catkin_make_isolated` to solve this.

pibaopi, 2021-12-13:
Did you solve it? I ran into the same problem on melodic

Enes1097, 2021-12-13:
> I just built the melodic-devel branch successfully. I am on Ubuntu 20.04 running ROS1 Noetic. Your build is not actually failing; CMake is quitting at the configuration stage, so it never attempts a [...]

> The error message literally says:
> 
> > This workspace contains non-catkin packages in it, and catkin cannot build a non-homogeneous workspace without isolation. Try the `catkin_make_isolated` command instead.
> 
> So use `catkin_make_isolated` to solve this.

As much as I remember, I actually tried using catkin_make_isolated which did not solve this problem.

> Did you solve it? I ran into the same problem on melodic

No, I sadly did not solve it. The last time I tried this was on that day in September. I just got around it using a Matlab Tool.

closed 2021-12-13 by Enes1097

pibaopi, 2021-12-13:
@Enes1097 Well, I also compiled successfully through another method in the first two hours!

ozehken, 2022-10-10:
> @Enes1097 Well, I also compiled successfully through another method in the first two hours!

I have same problem. How did you solve it?